bespoke-silicon-group/basejump_stl #3: 1RW sync bitmask does not build in verilator

dpetrisko, 2019-04-04:
When you try to compile bsg_mem/bsg_mem_1rw_sync_mask_write_bit_synth.v with Verilator, you get this error:

%Error-BLKLOOPINIT: /home/petrisko/bitbucket/test/pre-alpha-release/basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_bit_synth.v:42: Unsupported: Delayed assignment to array inside for loops (non-delayed is ok - see docs)

Verilator does not like this construct
```
 always_ff @(posedge clk_i)
   if (v_i & w_i)
       for (i = 0; i < width_p; i=i+1)
         if (w_mask_i[i])
           mem[addr_i][i] <= data_i[i];
```

This would work functionally, but may not synthesize to the same, depending on compiler quality.

```
 always_ff @(posedge clk_i)
   if (v_i & w_i)
     mem[addr_i] <= (~wmask_i & mem[addr_i]) | (w_mask_i & data_i); 
```

dpetrisko, 2019-04-04:
This is a fairly high priority fix because it breaks BlackParrot's Verilator build
@proftaylor

dpetrisko, 2019-04-04:
Oh, actually this could probably be fixed by changing to mem_n semantics.  I'll experiment

closed 2019-04-04 by dpetrisko

black-parrot, 2019-04-04:
What was the resolution?

dpetrisko, 2019-04-04:
Made a PR to fix

stdavids, 2019-04-04:
PR#4