bespoke-silicon-group/basejump_stl #5: [QUALITY] PPA regression

black-parrot, 2019-04-05:
We should add a PPA regression to track PPA across synthesis tools, versions and pull requests.

stdavids, 2019-04-05:
Using dimensionless lib file (or the similar product from nand gate) we could even do some of this as part of CI.

drichmond, 2019-04-05:
Would be useful in the xilinx tools too - but harder.

black-parrot, 2019-04-05:
@dpetrisko do we just provide a VM image to use to run CI? Something we can bury a VCS license server pointer in?

dpetrisko, 2019-04-05:
So there are basically two ways to run CI.

1) we have a machine which has VCS installed, license variables configured, etc.  We also install a "runner" program which GitHub pings upon push.  The machine then clones the repo and runs tests.

2) create a Docker image, which has VCS installed.  We also install a "runner" program on a server (which is not required to have anything except Docker).  Upon a GitHub push, the "runner" downloads the Docker image, starts it up, clones the repo and runs tests.  The "runner" can also set any license variables needed when running the Docker image.

For a Docker image, we just have to put the Dockerfile (list of all steps to create the image) in the repo for everyone to use it.  You can also publish Docker images, but they tend to be large (one with fairly minimal Verilator and BlackParrot was 800 MB, if we include VCS and DC it'll probably be 10GB).  There also are probably some legal restrictions on what you can include in a Dockerfile/image.

For BlackParrot, the current thinking is to include a Dockerfile for the FOSS toolchain to run a primitive regression and have the CI run baremetal for VCS and DC.  I think a similar approach should work for BaseJump STL

dpetrisko, 2019-04-05:
(baremetal meaning on a dedicated AWS CI instance, due to PDK restrictions.  We'll need Scott's dimensionless lib file to do this)

drichmond, 2019-04-05:
Would it make sense to have a CI server locally?

dpetrisko, 2019-04-05:
We can't have a CI server on the same machine as our NDA protected PDKs.  But we could buy a small server to run the CI/CAD tools, which would be convenient, not having to worry about turning on/off AWS instances.

stdavids, 2019-04-05:
@drichmond What challenges do you foresee with the Xilinx tools performing a PPA regression test?

drichmond, 2019-04-05:
It’s doable - but tricky. The vivado tools just aren’t set up to deal with circuits that aren’t complete systems. You have to tell the tool that your pins are virtual IOs and shouldn’t be placed, or define partial reconfiguration region where the pins are at the interface. Otherwise the tools optimize your circuit away. 

I can look into this again though. It’s possible there’s an easier way than I recall 

This is one reason why RIFFA was impactful - for years people had been submitting synthesis results for their architectures. This isn’t accurate by any measure. RIFFA made it easier to integrate your accelerator into a system and get actual PPA results for an FPGA.

black-parrot, 2019-04-05:
We could maybe just run synthesis for FPGA to start?