bespoke-silicon-group/basejump_stl #17: [BUG]

taylor-bsg, 2019-04-11:
https://github.com/bespoke-silicon-group/basejump_stl/blob/180c9648fc061dbe6e2eff4ab2909724effe39c9/bsg_mem/bsg_mem_2r1w_sync.v#L75

Assertions should be triggered in negedge of clock to avoid races.

taylor-bsg, 2019-04-11:
https://github.com/bespoke-silicon-group/basejump_stl/blob/180c9648fc061dbe6e2eff4ab2909724effe39c9/bsg_mem/bsg_mem_1r1w_sync.v#L73

dpetrisko, 2019-04-11:
I've heard this before, but never understood.  Since assertions are nonsynth, what is the race?

taylor-bsg, 2019-04-11:
Verilog simulations have various algorithms for evaluating statement
ordering and they don’t always get it right if your checks or your input
vectors are triggered on the same edge of the clock as the flops that drive
or receive them. This kind of issue is why you used to see stuff like:

a <= #1 b;

closed 2021-08-06 by taylor-bsg