bespoke-silicon-group/basejump_stl #24: [CLARITY] for single bit wires; replace logic foo; assign foo = XX; with wire foo = XX;

taylor-bsg, 2019-04-19:
https://github.com/bespoke-silicon-group/basejump_stl/blob/57a74de8023234987cc1e58311d10878ddc6f96d/bsg_cache/bsg_cache.v#L60

dpetrisko, 2019-04-19:
Why stop at single bit wires? 

wire [1:0] bar = XX works as well

taylor-bsg, 2019-04-19:
Agreed; I just wrote this because for the code I was pointing at it is
mostly single bit wires. The important thing is not do to this for
structures, or things that you declare as logic, i..e logic foo = XX is bad.

M

dpetrisko, 2019-04-19:
Think that (wire =) should go in the style guideline?

taylor-bsg, 2019-04-19:
The guide mentioned you could do this, but did not proactively suggest
doing it -- I added it! Thank you.

M

tommydcjung, 2019-04-19:
might be a good idea to create a module bsg_cache_pkt_decode. and not rely on the actual bit values of enum.

taylor-bsg, 2019-04-19:
Maybe post a separate issue for this ..

closed 2019-05-11 by tommydcjung