bespoke-silicon-group/basejump_stl #35: Test for bsg_mem_banked_crossbar appears to be out-dated/broken

dhanna11, 2019-05-15:
Hi, I'm trying to evaluate the bsg_mem_banked_crossbar for a project I'm working on.
I can't seem to get the test to work correctly. 
Using modelsim 2015, I only get 
> Error: Error at time:             (time), no transaction in a cycle

I changed https://github.com/bespoke-silicon-group/basejump_stl/blob/82873839281e09a1d36e3f586e76f7aaeb49320d/testing/bsg_mem/bsg_mem_banked_crossbar/test_bsg.v#L151-L166

to 
```
bsg_mem_banked_crossbar #( .bank_size_p  (bank_size_lp)
                            ,.num_ports_p  (ports_lp)
                            ,.num_banks_p  (banks_lp)
                            ,.rr_lo_hi_p(0) 
                            ,.data_width_p (data_width_lp)
                           ) UUT
                           ( .clk_i   (clk)
                            ,.reset_i (reset)
                            ,.reverse_pr_i('0)
                            ,.v_i     (test_input_v)
                            ,.w_i     (test_input_w)
                            ,.addr_i  (test_input_addr)
                            ,.data_i  (test_input_data)
                            ,.mask_i  (test_input_mask)
                            ,.yumi_o  (test_output_yumi)
                            ,.v_o     (test_output_v)
                            ,.data_o  (test_output_data)
                           );
```
and messed around with the makefile to get vsim to stop complaining about missing files:
https://github.com/bespoke-silicon-group/basejump_stl/blob/82873839281e09a1d36e3f586e76f7aaeb49320d/testing/bsg_mem/bsg_mem_banked_crossbar/Makefile#L17 to 
```
BSG_MISC_FILES      =   bsg_defines.v bsg_crossbar_o_by_i.v bsg_transpose.v bsg_cycle_counter.v bsg_encode_one_hot.v bsg_mux_one_hot.v bsg_round_robin_arb.v
```
and 
https://github.com/bespoke-silicon-group/basejump_stl/blob/82873839281e09a1d36e3f586e76f7aaeb49320d/testing/bsg_mem/bsg_mem_banked_crossbar/Makefile#L23
to 
```
BSG_MEM_FILES       =   bsg_mem_1rw_sync.v bsg_mem_1rw_sync_synth.v bsg_mem_1rw_sync_mask_write_byte.v bsg_mem_1rw_sync_mask_write_byte_synth.v
```

taylor-bsg, 2019-05-15:
Thanks for the bug report! We will take a look. The contributor who wrote the test used modelsim, which we do not use here; we might look into mapping to FuseSoc which would allow greater portability.

taylor-bsg, 2019-05-16:
@dhanna11 Should be fixed now -- give it a try. Thanks again!

closed 2019-05-16 by taylor-bsg