bespoke-silicon-group/basejump_stl #36: Port testing directory to make use of FuseSOC

taylor-bsg, 2019-05-15:
This will allow us to portably build the tests for different verilog simulators so that people can more easily use the tests.

mithro, 2019-05-16:
FYI - @olofk

olofk, 2019-05-16:
Happy to help out here. Just let me know if you have any questions or seek guidance

taylor-bsg, 2019-05-20:
Thanks Olof!

Any suggestions for what components of BaseJump STL should be FuseSOC'd?

It seems like there are two modalities. The first is people using BaseJump
STL as a library. The second is people running the regression test on their
infrastructure. Currently each test has a configuration for a different
tool flow, depending on what the tester had. Of course it would be neat to
have FuseSOC abstract that away..

Of course, if you have any cycles to do a proof of concept of setup for our
students to follow, that also would be appreciated!

M