bespoke-silicon-group/basejump_stl #40: [FEATURE] Add ability to lock cache lines, to mitigate hard errors or optimize for realtime.

taylor-bsg, 2019-05-17:
https://github.com/bespoke-silicon-group/basejump_stl/blob/8449973b2a23f1889f8a76da1d7e5cb48713f870/bsg_cache/bsg_cache.v#L2

tommydcjung, 2019-05-17:
what does it mean to have cache line "locked"?

taylor-bsg, 2019-05-17:
As discussed in the tapeout team meeting...

dpetrisko, 2019-05-17:
"Data cache locking for higher program predictability" - Vera, et al.
http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.13.2239&rep=rep1&type=pdf

Essentially, the line needs to be ineligible for eviction. This will require a lock bit in the stat for the cache line and modification of the LRU algorithm to ignore these lines.  Thus, if there is a tag match, we still do loads and stores, but we never evict.

I think we should only be able to lock half of the ways, because otherwise the LRU algorithm could find no eligible way (it also could become critical-pathy).

Question for BlackParrot cache: how should it respond to coherence transactions while in a locked state?

tommydcjung, 2019-05-17:
what if the line is locked and invalid, can it refill a new line there?

dpetrisko, 2019-05-17:
No, because we may be locking it because the line has hard faults.

dpetrisko, 2019-05-17:
(In which case we will lock it with an invalid tag)

tommydcjung, 2019-05-17:
what if both ways were defective? are we assuming that that is unlikely to happen?

dpetrisko, 2019-05-17:
Yeah, the chip's just dead at that point. We could make it so that you bypass the cache if you map to a dead set, but let's punt that until later

tommydcjung, 2019-05-17:
It's possible to bypass, but we would need to mask bits to the cache dma interface.

taylor-bsg, 2019-05-17:
Not worth the complexity.

M