bespoke-silicon-group/basejump_stl #45: Should use bsg_cadenv to set up tests

dpetrisko, 2019-05-19:
Right now, if you run many of the tests in testing (testing/bsg_noc/bsg_wormhole_router_adapter_in), they expect you to have vcs on your PATH, along with license servers set up, etc. 

We should have a public bsg_cadenv repo where users can set up the CAD variables needed and then Makefiles in BaseJump STL should use the correct variables and make no assumptions about environment.

This could be a shorter term solution to #36, or possibly orthogonal. I don't know how license servers work with FuseSoC

dpetrisko, 2019-05-20:
Regardless of that, there's also a dependency on $BSG_IP_CORES_DIR

taylor-bsg, 2019-05-20:
Yeah they are all inconsistent — just whatever the contributor had on hand.
We want to use FuseSoc so we can verify across tool chains and also so that
end users can easily move across to their own tools.