bespoke-silicon-group/basejump_stl #53: [bsg_fpu] bsg_fpu_cmp make pipeline_p parameter

tommydcjung, 2019-05-23:
pipeline_p=1 : 1-stage pipeline
pipeline_p=0 : no pipeline

taylor-bsg, 2019-05-28:
Probably should just have no pipeline? The critical path is very short and who is to see whether the user wants the register before or after the logic.

tommydcjung, 2019-05-28:
I think I created an issue with wrong title. I meant to say bsg_fpu_f2i. 
I removed the pipeline to fit it in int-float pipeline.
The critical path for f2i is when we are converting floating to two's complement negative int, we have to add 1 at the end. This might be okay, since f2i does not round, but instead truncate.

taylor-bsg, 2019-05-29:
Okay we will see if it is critical in the design. It probably was not a
path we optimized in Raw.

taylor-bsg, 2026-09-11:
| 19 | Issue [#53](https://github.com/bespoke-silicon-group/basejump_stl/issues/53) — [bsg_fpu] bsg_fpu_cmp make pipeline_p parameter | Close — agreed outcome present | The discussion corrected the target to f2i and agreed to remove its pipeline. Current bsg_fpu_f2i has a combinational interface with no clock/pipeline parameter. [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/bsg_fpu/bsg_fpu_f2i.sv#L24). |

closed 2026-09-11 by taylor-bsg