bespoke-silicon-group/basejump_stl #84: More Documents is needed

WenhMa, 2019-06-13:
There are too many modules in the library. Sometimes a simple function can be implemented by two or more modules. I cannot find a doc to help me to choose the proper module. It is better to offer
1) A user manual to specify the interface and function of each module. It can also help the user to understand the differences between similar module.
2) A design specification to explain the principle of each module. It maybe useful for senior designer.

taylor-bsg, 2019-06-13:
Thank you for the feedback WenhMa -- I think documentation is a plus and we are working on it (but also looking for contributors.)

WenhMa, 2019-06-13:
Would you please show the demo that you are working on? Maybe I and my classmates could contribute to this.

taylor-bsg, 2026-09-11:
| 29 | Issue [#84](https://github.com/bespoke-silicon-group/basejump_stl/issues/84) — More Documents is needed | Close — duplicate documentation request | Consolidate this general request into the API documentation umbrella #42, preserving the request for accessible introductory material. [issue #42](https://github.com/bespoke-silicon-group/basejump_stl/issues/42). |

closed 2026-09-11 by taylor-bsg