bespoke-silicon-group/basejump_stl #88: bsg_fifo_multiplexed

taylor-bsg, 2019-06-15:
This would be a multiplexed FIFO implementation, which leverages a single hardened SRAM to store state. The idea is that on input, you have a tag which is which channel it corresponds to.  And somehow you have the ability to pull data based on what channel you want to access. 

A lot of different implementations, TBD what the right interface is.

dpetrisko, 2019-08-15:
Proposal:

```
bsg_fifo_multiplexed
 #(parameter width_p          = "inv"
     , parameter els_p             = "inv"
     , parameter tag_width_p = "inv"
     )
   (input clk_i
    , input reset_i

    , input [tag_width_p-1:0] input_tag_i
    , input [width_p-1:0] data_i
    , input v_i
    , input yumi_o

    , input [tag_width_p-1:0] output_tag_i
    , output [width_p-1:0] data_o
    , output v_o
    , input ready_i
    )
```

Notice the inversion of the typical FIFO handshake.  
```
v_i & (we have space in input_tag_i channel) -> yumi_o
ready_i & (we have valid entry in output_tag_i channel) -> v_o
```

taylor-bsg, 2019-08-15:
Thought provoking idea! I think maybe we wait until we need this module and
then we will know what semantics it should have.

An alternative would be to give it N ready lines on the input, so that we
can keep the same convention of non-shared FIFOs.

On the output it is more tricky. I think in most cases, for a multiplexed
channel, you don’t want to poll for availability. But there are a couple of
semantics — a) round-robin, give me the next item from any channel with its
channel #, b) give me this channel if there is anything, otherwise nothing,
or generalize c) round robin, give me the next item from this subset of
specified channels. d) tell me what is available, and I will tell you what
I want.  A can be done with a standard valid/yumi interface. B, C, and D
would need a cycle of delay between selecting the channel and the data
appearing in order to use a synchronous ram, which is a little irregular.
(There are also variants where you allow more than one word to be pulled
out of the multi FIFO per cycle...)

drichmond, 2019-08-19:
I think we had/have something like this in RIFFA. It was timing critical and super useful. 

I can look into what we did there and why if that information is useful.