bespoke-silicon-group/basejump_stl #96: Unguarded delay statement in bsg_wormhole_router_generalized

dpetrisko, 2019-06-19:
%Warning-STMTDLY: /home/petrisko/bitbucket/pre-alpha-release/basejump_stl/bsg_noc/bsg_wormhole_router_generalized.v:45: Unsupported: Ignoring delay on this delayed statement.
%Warning-STMTDLY: Use "/* verilator lint_off STMTDLY */" and lint_on around source to disable this message.

I think we need `ifndef VERILATOR surrounding this.  This will just get rid of the assertion in Verilator.  Otherwise, the assertion may fire only in verliator testbenches. Another solution would be to find a way to mask the assertion when not useful without delay (since delay is awful for many other reasons).

taylor-bsg, 2019-06-21:
Can add a reset !===1’b0 || and remove delay.

dpetrisko, 2019-07-17:
Accidentally created duplicate issue with more detail, Moved to #107

closed 2019-07-17 by dpetrisko