bespoke-silicon-group/basejump_stl #100: Add 1R1W "small fifo"

taylor-bsg, 2019-06-21:
Would use hardened 1R1W and a bsg_dff_en_bypass so that it can grab the read element out. Need to be careful about simultaneous read and write to same address of the 1R1W.

taylor-bsg, 2019-08-15:
Also makes sense to add an equivalent version for bsg_async_fifo

dpetrisko, 2019-08-15:
Does it make sense to have 'harden_p' as a parameter to the regular bsg_fifo_1r1w_small, since the changes should be fairly minimal?

taylor-bsg, 2019-08-16:
I think it could make sense. Hardening will require a synchronous SRAM which make the changes less minimal then expected though.

gaozihou, 2019-08-16:
Yeah I agree, we may want to slightly modify control logic to support bsg_mem_1r1w_sync.

One possible idea is that we add a port named **"rptr_n_o"** to **bsg_fifo_tracker** and **bsg_circular_ptr**. **"rptr_n_o"** is one cycle ahead of **"rptr_r_o"**, when attaching **"rptr_n_o"** to **bsg_mem_1r1w_sync**, the behavior will be same as before, latency of fifo will still be 1-cycle.

To deal with read-write-same-addr problem, as prof suggested, we may use a dff to hold the read data (just like what manycore regfile does).

taylor-bsg, 2019-08-16:
To preserve single-cycle latency, we also need to bypass the write data
into the read data register, disabling the read enable on the ram.

I’m not sure if there is enough similarity in the logic to justify
extending the existing fifo tracker module; it may be easier to create a
new module.  I guess we will see...

M

dpetrisko, 2020-09-30:
Close from #138?

closed 2021-08-06 by taylor-bsg