bespoke-silicon-group/basejump_stl #107: bsg_wormhole_router_generalized is not Verilator compatible

dpetrisko, 2019-07-17:
Fix 1: (bare #1000 statement).  Actual best practice for this is $assertoff ...delay... $asserton in the testbench, but this fix will work, too

```
`ifndef SYNTHESIS
`ifndef VERILATOR
    wire [dirs_lp-1:0][dirs_lp-1:0] matrix_out_in_transpose;

    bsg_transpose #(.width_p(dirs_lp),.els_p(dirs_lp)) tr (.i(routing_matrix_p[0])
                                                          ,.o(matrix_out_in_transpose)
                                                          );
    initial
      begin
        #1000;
        assert (routing_matrix_p[1] == matrix_out_in_transpose)
          else $error("inconsistent matrixes");
      end
`endif
`endif

```

But even with that, I get this error: 

[image]

dpetrisko, 2019-07-17:
Ah, there actually is a tristate expression in bsg_unconcentrate_static.  Is this necessary?  I thought 'z was considered bad practice nowadays.

If necessary, the fix is to `ifdef VERILATOR the tristate assignment.  Should it be x? 0?
If not, we should change the tristate to something else

taylor-bsg, 2019-07-23:
The solution is to add a ifdef VERILATOR to this module:

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_defines.v

For the BSG_DISCONNECTED_IN_SIM and BSG_UNDEFINED_IN_SIM macros.

Since verilator does not support 4-state simulation.

M

dpetrisko, 2019-07-23:
kk, I'll make a PR.  We only need it in BSG_DISCONNECTED_IN_SIM, though.  Verilator does have limited support of X's

taylor-bsg, 2019-07-23:
Oh I see — then we could use ‘X for disconnected in Verilator.

dpetrisko, 2019-08-03:
#122

closed 2019-08-15 by dpetrisko