bespoke-silicon-group/basejump_stl #113: System task causes a warning in Vivado

drichmond, 2019-07-25:
https://github.com/bespoke-silicon-group/basejump_stl/blob/d59e7bfef30abf44e1c13d48b286d92175c691e0/bsg_misc/bsg_dlatch.v#L10

Don't know if it's worth addressing - or if it can be - but here's the warning: 

```
WARNING: [Synth 8-1921] elaboration system task fatal violates IEEE 1800 syntax [/mnt/bsg/diskbits/dustinar/bsg_bladerunner/bsg_f1_621ec48/cl_manycore/build/src_post_encryption/bsg_dlatch.v:10]```

dpetrisko, 2019-07-25:
What Verilog standard does Vivado use?

dpetrisko, 2019-07-25:
https://forums.xilinx.com/t5/Synthesis/Elaboration-system-task-in-SystemVerilog/td-p/794022

It appears to be fixed in 2017.1. What Vivado version are you using?

drichmond, 2019-07-25:
2018.2

dpetrisko, 2019-07-25:
Sorry accidentally deleted instead of edited.

https://forums.xilinx.com/t5/Synthesis/Elaboration-system-task-in-SystemVerilog/td-p/794022

It's a Vivado bug, is there an ifdef VIVADO or something to wrap around?