bespoke-silicon-group/basejump_stl #115: RFC: Wrap $display statements in a DEBUG macro?

drichmond, 2019-07-25:
(x-post to BSG Manycoree)
It seems unnecessary to have $display statements enabled for every simulation run. Printing information on every run can slow the simulator down, or clog with unnecessary information. 

In BSG F1 we have a printing header. bsg_pr_debug statements are only enabled when the DEBUG macro is defined. This can be defined globally, or on a per-file basis. 

Instead of putting raw $display statements in BaseJump Code, it would be cool if we could wrap $display in a macro that is only enabled when DEBUG is defined. Like in C/C++ we could enable this on a per-file basis (by putting `define DEBUG above `include "bsg_basejump_printing.vh") or globally when compiling the simulation executable

dpetrisko, 2019-07-25:
Agree with the high level idea.  Global defines are pretty finicky with compile order and weird tool inconsistencies, so I would suggest using parameters for this.  This would allow us to enable on a per-module, per-file or global basis depending on how the parameters are set.

Here's an example of how to use parameters to enable debug to null, display, or file

https://github.com/black-parrot/pre-alpha-release/blob/master/bp_common/src/include/bp_common_defines.vh

drichmond, 2019-07-25:
I agree with your assessment - global defines (and defines in general) can be finicky. 

Is it possible to get the best of both worlds? Could we use bind, or a package, to set which modules to debug at a global scope?

taylor-bsg, 2019-07-25:
Generally we have a debug_p parameter that controls this information. Some
modules have different debug levels.

What kind of display statements are you referring to?

Generally we are friendly to one-time statements that print at the
beginning, in particular for things that change depending on the simulation
you are running. Things that fire and fire over and over we do not like to
have turned on long term.

Legitimate errors and warnings we generally want on all the time but can
add some code to silence during reset.

M

drichmond, 2019-07-25:
I'm referring to this in the Manycore (and a few other examples). These are useful -- for example Borna and I used them to debug -- but they're not clogging the simulation output for CUDA programs. My concern is that these statements are actually slowing down large scale simulation like they would in a C/C++ program.

https://github.com/bespoke-silicon-group/bsg_manycore/blob/64858dd07eaa4b7f8a767b89edc073947bb60bd6/v/vanilla_bean/network_rx.v#L330

taylor-bsg, 2019-07-26:
I see — you see these every time a CUDA kernel is launched?

drichmond, 2019-07-26:
CUDA, SPMD, etc. CUDA is just the worst, since it freezes/unfreezes tiles repeatedly

taylor-bsg, 2019-07-26:
I see — this used to be something that only happens once so it was not
disruptive, and it was helpful to know when a tile was supposed to start
running.  Solution might be to replace with val watcher from BaseJump STL
bsg_test, which allow you to say “trigger only once”.

M

drichmond, 2019-07-26:
That would be an acceptable solution - But there’s a good argument to have it display repeatedly too - that’s how Borna and I found a bug in our code during the tapeout

taylor-bsg, 2019-12-14:
In the context of BaseJump STL (and maybe bsg manycore), I am starting to thing that it makes sense to have a centralized package which controls the verbosity of printing of things. Then you go to one place to control verbosity. One goal would be to make to sure that people can use BaseJump STL files piecemeal.

dpetrisko, 2019-12-14:
The interface being something like `+define+BASEJUMP_LOG_LEVEL=DEBUG`
?

taylor-bsg, 2019-12-14:
More like a SystemVerilog package that allows you to control the kinds of
things that the simulator reports via variables or functions.

dpetrisko, 2019-12-14:
Oh, I see.  The UVM-style approach is to have both a “SEVERITY” which is associated with each print function and “VERBOSITY” which is a global, per build/run setting. 

http://cluelogic.com/2015/05/uvm-tutorial-for-candy-lovers-reporting-verbosity/

Could certainly imagine using parameters, macros, functions, whatever for something like this. 

> One goal would be to make to sure that people can use BaseJump STL files piecemeal.

What do you mean by this?

taylor-bsg, 2019-12-15:
i.e. you should be able to use some modules from BaseJump STL without
having to include a lot of unrelated stuff.