bespoke-silicon-group/basejump_stl #127: question about testing

edcote, 2019-08-08:
I was reading the [paper](http://cseweb.ucsd.edu/~mbtaylor/papers/Taylor_DAC_BaseJump_STL_2018.pdf) and noted the following ...

(9) Testing Suite. The STL shall have unit tests for each module.

Is [this](https://github.com/bespoke-silicon-group/basejump_stl/tree/master/testing) the testing suite?

... is the above assertion correct (for each module)?

taylor-bsg, 2019-08-09:
Hi,

Yes — those are the unit tests. Most of the modules are also used in our
SoC designs, so they are also tested regularly in silico.

M

NP95, 2019-09-05:
I have a query regarding the Basejump STL project. 

In the folder testing, those folders which are already present. Does that mean those tests are the complete unit tests for that IP and no more tests are required? For example bsg_cache in testing directory does that mean all those folders are tests to verify the bsg_cache IP and no additional tests are required?

dpetrisko, 2019-09-05:
Hi @NP95,

Certainly not. We have not formally verified these components (and even if we had, there’s a long discussion about how bugs creep in even with formal =)).

Additional unit tests are useful so long as they:

1) verify new or existing functionality (when a feature is added, or if the feature is currently untested)
2) verify new corner cases (when a bug is found, add a new test)
3) do not rely on specific module behavior to pass (test interfaces, not implementation)
4) do not require undue upkeep (trace_replay with a python generator is useful for this, bsg_cache being an excellent example)

We would love to accept PRs to testing/ for both modules which have no tests and modules which have some testing coverage.

closed 2021-08-06 by taylor-bsg