bespoke-silicon-group/basejump_stl #129: counter can not overflow in bsg_counter_overflow_en

leonardxiang, 2019-08-14:
The counter pointer in this module can not hit the value of power of 2.

leonardxiang, 2019-08-14:
Hello, @stdavids I create this issue here.

RicoLi424, 2023-09-07:
Hi Scott, I ran into the same issue as Leonard when setting `max_val_p` as 2^n. It seems the code has still not been updated to _`BSG_WIDTH`_ or `max_val_p+1` which can fix this issue. @stdavids

taylor-bsg, 2023-09-08:
Hi Rico,

You can submit a pull request

taylor-bsg, 2026-09-11:
| 4 | Issue [#129](https://github.com/bespoke-silicon-group/basejump_stl/issues/129) — counter can not overflow in bsg_counter_overflow_en | Close — fixed | The overflow counter now uses BSG_WIDTH(max_val_p), which allocates the extra bit when the maximum is a power of two. The original missing-overflow-bit case is addressed. [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/bsg_misc/bsg_counter_overflow_en.sv#L8); [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/bsg_misc/bsg_defines.sv#L50). |

closed 2026-09-11 by taylor-bsg