bespoke-silicon-group/basejump_stl #143: bsg_mux.v does not support els_p=1

tommydcjung, 2019-09-01:
if els_p=1, data_o should be data_i, regardless of sel_i.

taylor-bsg, 2019-09-02:
I believe it supports this..

dpetrisko, 2019-09-02:
> assign data_o = data_i[sel_i];

Only if you tie of sel_i. I think the behavior should be for els_p == 1, then sel could be left hanging

tommydcjung, 2019-09-02:
nope it doesn't.
[image]

taylor-bsg, 2019-09-03:
That is the correct behavior. sel_i of 1 is invalid if there is only one
input to the mux. It must be 0.

tommydcjung, 2019-09-03:
I think there is some advantage for having a special case for els_p=1.
Consider a mux in some design (e.g. bsg_cache) where the number of inputs (els_p) can vary from 1 to n, and the select bits come from a portion of address (e.g. `addr[0+:BSG_SAFE_CLOG(els_p)]`). Since we cannot have zero-width signal, we have to choose at least one bit from addr. This bit can be either one or zero, but bsg_mux outputs 'X for the input `1'b1`. I guess my argument is, for parameterizability of other modules, it is beneficial to have the property I mentioned.

Kind of like for bsg_decode. If num_out_p=1, `o` is always 1'b1, regardless of the input `i`.

dpetrisko, 2019-09-03:
I agree with Tommy for the default behavior.  

There is a compromise solution as well: 
unary_mux_passthrough_p=1 =>   o=i for sel=X
unary_mux_passthrough_p=0 =>  o=X for sel=1

taylor-bsg, 2019-09-03:
Seems reasonable for bsg_mux!

But why are you using that module in bsg_cache?  Maybe you should be using
bsg_mux_one_hot? ;-)

M

taylor-bsg, 2020-07-02:
bsg_mux now handles els_p=1 and ignores sel_i.

closed 2020-07-02 by taylor-bsg