bespoke-silicon-group/basejump_stl #152: bsg_clk_gen's testbench

duyhieubui, 2019-09-13:
I am trying to run bsg_clk_gen testbench with the behavior model by changing the makefile. However, it seems that test_bsg.v in testing/bsg_clk_gen is out of date.
This is the error message:

> Error-[UPIMI-E] Undefined port in module instantiation
> ../../bsg_clk_gen/bsg_clk_gen.v, 114
>   Port "bsg_tag_trigger_i" is not defined in module 'bsg_clk_gen_osc' defined 
>   in "../../bsg_clk_gen//bsg_clk_gen_osc.v", 14
>   Module instance: bsg_clk_gen_osc #(.num_adgs_p(num_adgs_p)) 
>   clk_gen_osc_inst( .bsg_tag_i (bsg_osc_tag_i),  .bsg_tag_trigger_i 
>   (bsg_osc_trigger_tag_i),  .async_reset_i (async_o ...
> 
> 
> Error-[UPIMI-E] Undefined port in module instantiation
> test_bsg.v, 35
>   Port "bsg_tag_clk_o" is not defined in module 'bsg_nonsynth_clk_gen_tester' 
>   defined in "bsg_nonsynth_clk_gen_tester.v", 59
>   Module instance: bsg_nonsynth_clk_gen_tester #(.num_adgs_p(bsg_num_adgs_lp),
>   .ds_width_p(bsg_ds_width_lp), .tag_els_p(bsg_tag_els_lp)) tester(.ext_clk_i,
>   .bsg_tag_clk_o, .bsg_ ...
> 
> 
> Lint-[TFIPC-L] Too few instance port connections
> test_bsg.v, 35
> "bsg_nonsynth_clk_gen_tester #(.num_adgs_p(bsg_num_adgs_lp), .ds_width_p(bsg_ds_width_lp), .tag_els_p(bsg_tag_els_lp)) tester(.bsg_tag_clk_o,  .ext_clk_i (ext_clk_i),  .bsg_tag_en_o (bsg_tag_en_o),  .bsg_tag_data_o (bsg_tag_data_o),  .bsg_clk_gen_sel_o (bsg_clk_gen_sel_o),  .bsg_clk_gen_async_reset_o (bsg_clk_gen_async_reset_o),  .bsg_clk_gen_i (bsg_clk_gen_i));"
>   The above instance has fewer port connections than the module definition,
>   input port 'bsg_tag_clk_i' is not connected,
>   input port 'start_i' is not connected,
>   output port 'done_o' is not connected.
> 
> 
> Lint-[TFIPC-L] Too few instance port connections
> test_bsg.v, 70
> "bsg_clk_gen #(.downsample_width_p(bsg_ds_width_lp), .num_adgs_p(bsg_num_adgs_lp)) DUT( .bsg_osc_tag_i (tags[0]),  .bsg_ds_tag_i (tags[1]),  .async_osc_reset_i (bsg_clk_gen_async_reset_o),  .select_i (bsg_clk_gen_sel_o),  .clk_o (bsg_clk_gen_i),  .ext_clk_i (ext_clk_i));"
>   The above instance has fewer port connections than the module definition,
>   input port 'bsg_osc_trigger_tag_i' is not connected.
>

taylor-bsg, 2019-09-13:
Thanks for your report! We usually use the hardened variants of those files (in the /hard directory) so this one may have gotten a little stale.

@stdavids can you take a lot and see if you can update it?

duyhieubui, 2019-09-15:
Thank you for the reply. Actually, we do not have access to the technologies in /hard. I have a plan to create the design with Synopsys Educational PDK 28-32nm and  TSMC 65GPlus. I believe a working behavior model will help a lot.

taylor-bsg, 2019-09-17:
I see — you should be able to do a gate-by-gate port of the 40nm clock
generator hard file to 65 in a few hours and get reasonable results and
something you could tapeout. We seldom use the behavioral model because it
is so easy to port.

M

taylor-bsg, 2019-09-17:
BTW it’s not the tester that is out of date, it is the behavioral model of
the clock generator. :-)

duyhieubui, 2019-09-17:
Thank you for the guideline. Following your guideline, I changed all the cell in tsmc_40 to the one in TSMC 65nm GPlus. TSMC 65nm GPlus's std cells  do not have the suffix "bwp" in the cell name, therefore, I removed this suffix and changed the Makefile for compilation. The design can be compiled with some modification to the makefile and test_bsg.v. However, when I ran it, it hung.
The reason might be some ports are not connected:
> Lint-[TFIPC-L] Too few instance port connections
> ../../bsg_clk_gen/bsg_clk_gen.v, 90
> "bsg_clk_gen_osc #(.num_adgs_p(num_adgs_p)) clk_gen_osc_inst( .bsg_tag_i (bsg_os
> c_tag_i),  .async_reset_i (async_osc_reset_i),  .clk_o (osc_clk_out));"
>   The above instance has fewer port connections than the module definition,
>   input port 'bsg_tag_trigger_i' is not connected.
> 
> 
> Lint-[TFIPC-L] Too few instance port connections
> test_bsg.v, 35
> "bsg_nonsynth_clk_gen_tester #(.num_adgs_p(bsg_num_adgs_lp), .ds_width_p(bsg_ds_
> width_lp), .tag_els_p(bsg_tag_els_lp)) tester( .ext_clk_i (ext_clk_i),  .bsg_tag
> _clk_i (bsg_tag_clk_i),  .bsg_tag_en_o (bsg_tag_en_o),  .bsg_tag_data_o (bsg_tag
> _data_o),  .bsg_clk_gen_sel_o (bsg_clk_gen_sel_o),  .bsg_clk_gen_async_reset_o (
> bsg_clk_gen_async_reset_o),  .bsg_clk_gen_i (bsg_clk_gen_i));"
>   The above instance has fewer port connections than the module definition,
>   input port 'start_i' is not connected,
>   output port 'done_o' is not connected.
>

duyhieubui, 2019-09-17:
With these changes, I can start simulation but it still hangs
```

diff --git a/bsg_clk_gen/bsg_clk_gen.v b/bsg_clk_gen/bsg_clk_gen.v
index f595df2..8a6c397 100644
--- a/bsg_clk_gen/bsg_clk_gen.v
+++ b/bsg_clk_gen/bsg_clk_gen.v
@@ -90,6 +90,7 @@ module bsg_clk_gen
   bsg_clk_gen_osc #(.num_adgs_p(num_adgs_p))  clk_gen_osc_inst
     (
      .bsg_tag_i          (bsg_osc_tag_i    )
+     ,.bsg_tag_trigger_i          (bsg_ds_tag_i    )
      ,.async_reset_i     (async_osc_reset_i)
      ,.clk_o             (osc_clk_out      )
     );
diff --git a/testing/bsg_clk_gen/test_bsg.v b/testing/bsg_clk_gen/test_bsg.v
index 3925d75..afe6461 100644
--- a/testing/bsg_clk_gen/test_bsg.v
+++ b/testing/bsg_clk_gen/test_bsg.v
@@ -18,22 +18,25 @@ module test_bsg;
 
    `declare_bsg_tag_header_s(bsg_tag_els_lp,lg_max_payload_length_lp)
 
-   wire clk_i, reset_i, bsg_tag_clk_o, bsg_tag_en_o, bsg_tag_data_o;
+   wire bsg_tag_clk_i, bsg_tag_en_o, bsg_tag_data_o;
 
    wire [1:0] bsg_clk_gen_sel_o;
    wire       bsg_clk_gen_external_clk_o;
    wire       bsg_clk_gen_async_reset_o;
    wire       bsg_clk_gen_i;
    wire       ext_clk_i;
+   reg       start_i;
+   wire       done_o;
 
    bsg_nonsynth_clock_gen #(5ns) cfg_clk_gen (ext_clk_i);
+   bsg_nonsynth_clock_gen #(100ns) cfg_tag_clk_gen (bsg_tag_clk_i);
 
    bsg_nonsynth_clk_gen_tester #(.num_adgs_p(bsg_num_adgs_lp)
                                  ,.ds_width_p(bsg_ds_width_lp)
                                  ,.tag_els_p(bsg_tag_els_lp)
                                  ) tester
      (.ext_clk_i
-      ,.bsg_tag_clk_o
+      ,.bsg_tag_clk_i
       ,.bsg_tag_en_o
       ,.bsg_tag_data_o
 
@@ -41,6 +44,8 @@ module test_bsg;
       ,.bsg_clk_gen_async_reset_o
 
       ,.bsg_clk_gen_i
+	, .start_i
+	, .done_o
       );
 
   // Enable VPD dump file
@@ -49,6 +54,8 @@ module test_bsg;
     begin
       $vcdpluson;
       $vcdplusmemon;
+      start_i = 1'b0;
+      #500 start_i = 1'b1;
     end
 
 
@@ -60,7 +67,7 @@ module test_bsg;
    bsg_tag_master #(.els_p(bsg_tag_els_lp)
                     ,.lg_width_p(lg_max_payload_length_lp)
                     ) btm
-     (.clk_i       (bsg_tag_clk_o)
+     (.clk_i       (bsg_tag_clk_i)
       ,.data_i     (bsg_tag_data_o)
       ,.en_i       (bsg_tag_en_o)
       ,.clients_r_o(tags)
```
It hangs at this message:
```
## INFO 1: performing async reset on oscillator (bsg_clk_gen_sel_o=0)
## INFO 2: testing for live oscillator (bsg_clk_gen_sel_o=0)
## clock_watcher {                                                                                POSEDGE offset (after 0        cycl
es) 4000    ps (n/p phase ratio=-0.000)} (test_bsg.tester.wtch)
## clock_watcher { NEGEDGE offset (after 0       cycles) 4000    ps (p/n phase ratio=1.000)} (test_bsg.tester.wtch)
## PASS 0: Counted 10 clock positive edges (bsg_clk_gen_sel_o=0)
## INFO 3b: beginning bsg_tag master reset transmit 
## INFO 3e: end bsg_tag master reset transmit
## INFO 4b: begin bsg_tag_client reset transmit for oscillator
## INFO 4e: end bsg_tag_client reset transmit for oscillator 
## INFO 5: attaching recv side of nodes to bsg_tag (bsg_tag_en_o=1)
## INFO 6: testing for live oscillator (bsg_clk_gen_sel_o=00)

```

taylor-bsg, 2019-10-04:
Sorry for the delayed response. Looking at your bsg_clk_gen/bsg_clk_gen.v,
it seems like it is out of date, because that line is commented out in
master. What branch are you pulling from?

M

duyhieubui, 2019-10-05:
Sorry for the confusion. I tried to find a working version by looping through the commit. The code that I modified is coming from the commit: c54f75c94ef12e9cb2eb0aba723ab7675dc1ce76 in which you said it is to support V2.

I will try to work from the master branch.

duyhieubui, 2019-10-05:
I tried with the latest code in the master branch. bsg_clk_gen_osc's interface is actually for the v1 model which only exists with the hard file of TSMC_180_250. In the following patch, I change the interface of bsg_clk_gen_osc to match with the v2. Some modifications for test_bsg is also included.

This patch might be a working testbench for the behavior model:

```
diff --git a/bsg_clk_gen/bsg_clk_gen.v b/bsg_clk_gen/bsg_clk_gen.v
index 4e703b6..7159892 100644
--- a/bsg_clk_gen/bsg_clk_gen.v
+++ b/bsg_clk_gen/bsg_clk_gen.v
@@ -113,8 +113,7 @@ module bsg_clk_gen
  */
        bsg_clk_gen_osc #(.num_adgs_p(num_adgs_p))  clk_gen_osc_inst
          (
-          .bsg_tag_i          (bsg_osc_tag_i        )
-          ,.bsg_tag_trigger_i  (bsg_osc_trigger_tag_i)
+          .bsg_tag_trigger_i (bsg_osc_trigger_tag_i)
           ,.async_reset_i     (async_osc_reset_i    )
           ,.clk_o             (osc_clk_out          )
           );
diff --git a/bsg_clk_gen/bsg_clk_gen_osc.v b/bsg_clk_gen/bsg_clk_gen_osc.v
index d1f5e9a..0dde705 100644
--- a/bsg_clk_gen/bsg_clk_gen_osc.v
+++ b/bsg_clk_gen/bsg_clk_gen_osc.v
@@ -17,7 +17,7 @@ module bsg_clk_gen_osc
     #(parameter num_adgs_p=1)
   (
    input async_reset_i
-   ,input bsg_tag_s bsg_tag_i
+   ,input bsg_tag_s bsg_tag_trigger_i
    ,output logic clk_o
    );
 
@@ -33,7 +33,7 @@ module bsg_clk_gen_osc
                     ,.harden_p(1)
                     ,.default_p(0)
                     ) btc
-     (.bsg_tag_i     (bsg_tag_i)
+     (.bsg_tag_i     (bsg_tag_trigger_i)
       ,.recv_clk_i   (clk_o)
       ,.recv_reset_i (1'b0)     // no default value is loaded;
       ,.recv_new_r_o (fb_we_r)  // default is already in OSC flops
diff --git a/testing/bsg_clk_gen/test_bsg.v b/testing/bsg_clk_gen/test_bsg.v
index bbb38b7..be6d7d5 100644
--- a/testing/bsg_clk_gen/test_bsg.v
+++ b/testing/bsg_clk_gen/test_bsg.v
@@ -9,7 +9,7 @@ module test_bsg;
 
    localparam bsg_num_adgs_lp  = `NUM_ADGS_P;
    localparam bsg_ds_width_lp  = `WIDTH_P;
-   localparam bsg_tag_els_lp  = 2;
+   localparam bsg_tag_els_lp  = 3;
 
    `declare_bsg_clk_gen_osc_tag_payload_s(bsg_num_adgs_lp)
    `declare_bsg_clk_gen_ds_tag_payload_s(bsg_ds_width_lp)
@@ -19,22 +19,25 @@ module test_bsg;
 
    `declare_bsg_tag_header_s(bsg_tag_els_lp,lg_max_payload_length_lp)
 
-   wire clk_i, reset_i, bsg_tag_clk_o, bsg_tag_en_o, bsg_tag_data_o;
+   wire bsg_tag_clk_i, bsg_tag_en_o, bsg_tag_data_o;
 
    wire [1:0] bsg_clk_gen_sel_o;
    wire       bsg_clk_gen_external_clk_o;
    wire       bsg_clk_gen_async_reset_o;
    wire       bsg_clk_gen_i;
    wire       ext_clk_i;
+   reg        start_i;
+   wire       done_o;
 
    bsg_nonsynth_clock_gen #(5ns) cfg_clk_gen (ext_clk_i);
+   bsg_nonsynth_clock_gen #(8ns) cfg_tag_clk_gen (bsg_tag_clk_i);
 
    bsg_nonsynth_clk_gen_tester #(.num_adgs_p(bsg_num_adgs_lp)
                                  ,.ds_width_p(bsg_ds_width_lp)
                                  ,.tag_els_p(bsg_tag_els_lp)
                                  ) tester
      (.ext_clk_i
-      ,.bsg_tag_clk_o
+      ,.bsg_tag_clk_i
       ,.bsg_tag_en_o
       ,.bsg_tag_data_o
 
@@ -42,6 +45,8 @@ module test_bsg;
       ,.bsg_clk_gen_async_reset_o
 
       ,.bsg_clk_gen_i
+      ,.start_i
+      ,.done_o
       );
 
   // Enable VPD dump file
@@ -50,18 +55,22 @@ module test_bsg;
     begin
       $vcdpluson;
       $vcdplusmemon;
+      start_i = 1'b0;
+      #500ns start_i = 1'b1;
+      @(done_o == 1'b1);
+      $finish(2);
     end
 
 
   // Clock generator signals
   //
 
-   bsg_tag_s [1:0] tags;
+   bsg_tag_s [2:0] tags;
 
    bsg_tag_master #(.els_p(bsg_tag_els_lp)
                     ,.lg_width_p(lg_max_payload_length_lp)
                     ) btm
-     (.clk_i       (bsg_tag_clk_o)
+     (.clk_i       (bsg_tag_clk_i)
       ,.data_i     (bsg_tag_data_o)
       ,.en_i       (bsg_tag_en_o)
       ,.clients_r_o(tags)
@@ -70,6 +79,7 @@ module test_bsg;
   bsg_clk_gen #(.downsample_width_p(bsg_ds_width_lp), .num_adgs_p(bsg_num_adgs_lp)) DUT
     (.bsg_osc_tag_i(tags[0])
      ,.bsg_ds_tag_i(tags[1])
+     ,.bsg_osc_trigger_tag_i(tags[2])
      ,.async_osc_reset_i(bsg_clk_gen_async_reset_o)
 
     ,.ext_clk_i
```

taylor-bsg, 2019-10-05:
Ah neat — were you able to get the updated purely behavioral model work in
simulation? Yeah generally the expectation would be that if you are using
hardened files, you would be using all of the hardened files from that
directory (e.g. TSMC 40’s bsg_clk_gen_osc.)

duyhieubui, 2019-10-06:
I tried with the behavior model and it can run the simulation smoothly. However, with the harden files, after async_reset it stucked.
Now, I understand more about your design. In general, the behavior model of bsg_clk_gen_osc is quite similar to v1 model. In v2, to handle some bugs, you introduce another tag client to handle the reset of the oscillator.
My current hack is to make bsg_clk_gen_osc have the same interface as the one in the hard files. Behavior model don't need to reset the oscillator in a specific way, therefore, I just ignored bsg_tag_i signal.

In my current simulation, I only use the osc, adt, cdt and fdt as the hard files (modified from tsmc_40 into tsmc_65gp), all the other files are still behavior models. During simulation, some important signals are 'X' therefore, the oscillator was stopped during the second phase at the step 6 of the testbench:
```
## INFO 4b: begin bsg_tag_client reset transmit for oscillator
## bsg_tag_master STATE CHANGE  # eStart --> eHeader #
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 0) = 1
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 1) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 2) = 1
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 3) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 4) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 5) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 6) = 0
## bsg_tag_master PACKET HEADER RECEIVED (length=0101,data_not_reset=0,nodeID=00
) (test_bsg.btm) 
## bsg_tag_master STATE CHANGE  # eHeader --> eTransfer #
## bsg_tag_master PACKET END (test_bsg.btm)
## bsg_tag_master STATE CHANGE  # eTransfer --> eStart #
## INFO 4e: end bsg_tag_client reset transmit for oscillator 
## INFO 5: attaching recv side of nodes to bsg_tag (bsg_tag_en_o=1)
## INFO 6: testing for live oscillator (bsg_clk_gen_sel_o=00)
```
After transmitting the reset to the oscillator, some signals becomes 'X', and the oscillator output is also 'X' and the simulation stop.

duyhieubui, 2019-10-22:
I finally found out my problem. I must set the testbench version to version 2 (by default it is version 1). Now I can run the simulation with all the hard files for TSMC 65nm. However, at the end of the simulation, I got the following errors in bsg_strobe.v. I do not know if it would be a problem.

```
* test_bsg.tester 
***********************************************************
*                                                         *
*                 CLOCK GEN TESTER FINISHED               *
*                                                         *
***********************************************************
                                                           
## Setting clk mux final value: 00 (test_bsg.tester).
## Setting final osc value: 00000000000000000000000000000000 (test_bsg.tester)
## clock_watcher { NEGEDGE offset (after 2       cycles) 4       ps (p/n phase ratio=8.000)} (test_bsg.tester.wtch)
## clock_watcher {                                                                                POSEDGE offset (after 2        cycles) 4       ps (n/p phase ratio=1.000)} (test_bsg.tester.wtch)
## bsg_tag_master STATE CHANGE  # eStart --> eHeader #
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 0) = 1
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 1) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 2) = 1
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 3) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 4) = 1
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 5) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 6) = 0
## bsg_tag_master PACKET HEADER RECEIVED (length=0101,data_not_reset=1,nodeID=00) (test_bsg.btm) 
## bsg_tag_master STATE CHANGE  # eHeader --> eTransfer #
## bsg_tag_master PACKET END (test_bsg.btm)
## bsg_tag_master STATE CHANGE  # eTransfer --> eStart #
## bsg_tag_master STATE CHANGE  # eStart --> eHeader #
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 0) = 1
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 1) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 2) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 3) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 4) = 1
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 5) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 6) = 1
## bsg_tag_master PACKET HEADER RECEIVED (length=0001,data_not_reset=1,nodeID=10) (test_bsg.btm) 
## bsg_tag_master STATE CHANGE  # eHeader --> eTransfer #
## bsg_tag_master PACKET END (test_bsg.btm)
## bsg_tag_master STATE CHANGE  # eTransfer --> eStart #
## bsg_tag_master STATE CHANGE  # eStart --> eHeader #
## bsg_tag_client (send) SHIFTING  0 (test_bsg.DUT.clk_gen_osc_inst.btc_trigger)
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 0) = 1
"../../bsg_misc/bsg_strobe.v", 139: test_bsg.DUT.clk_gen_ds_inst.strobe.unnamed$$_0: started at 34650000ps failed at 34650000ps
        Offending '((strobe_n === 'x) || (strobe_n == (&((C_r << 1) ^ S_r))))'
Error: "../../bsg_misc/bsg_strobe.v", 139: test_bsg.DUT.clk_gen_ds_inst.strobe.unnamed$$_0: at time 34650000 ps
## faulty assumption about strobe signal in test_bsg.DUT.clk_gen_ds_inst.strobe (C_r=0000000, S_r=11111100, strobe_n=1)
## clock_watcher { NEGEDGE offset (after 66      cycles) 0       ps (p/n phase ratio=inf)} (test_bsg.tester.wtch)
## clock_watcher {                                                                                POSEDGE offset (after 66       cycles) 0       ps (n/p phase ratio=-nan)} (test_bsg.tester.wtch)
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 1) = 0
## clock_watcher { NEGEDGE offset (after 1       cycles) 4       ps (p/n phase ratio=0.000)} (test_bsg.tester.wtch)
## clock_watcher {                                                                                POSEDGE offset (after 1        cycles) 4       ps (n/p phase ratio=1.000)} (test_bsg.tester.wtch)
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 2) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 3) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 4) = 1
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 5) = 0
## bsg_tag_master RECEIVING HEADER (test_bsg.btm) ( 6) = 1
## bsg_tag_master PACKET HEADER RECEIVED (length=0001,data_not_reset=1,nodeID=10) (test_bsg.btm) 
## bsg_tag_master STATE CHANGE  # eHeader --> eTransfer #
## bsg_tag_master PACKET END (test_bsg.btm)
## bsg_tag_master STATE CHANGE  # eTransfer --> eStart #
## bsg_tag_client (send) SHIFTING  1 (test_bsg.DUT.clk_gen_osc_inst.btc_trigger)
## Setting final downsampler to 00000000000000000000000000000000 (test_bsg.tester)
```

taylor-bsg, 2019-10-22:
Neat — can you include the full log?

duyhieubui, 2019-10-22:
This is the full simulation log file.

[run.log.zip](https://github.com/bespoke-silicon-group/basejump_stl/files/3755637/run.log.zip)

duyhieubui, 2019-10-24:
I committed the support for TSMC 65nm gplus here: https://github.com/duyhieubui/basejump_stl/commit/0c4ab878f3b2c57f0fb9ebc1d838aed9b470883a

duyhieubui, 2019-11-01:
During the simulation, there is one flip-flop which is not flipped at the time of the assertion. I am not sure if it is a problem of the simulation tools (might be fixed with some options) or it is because of the standard cell model.
[flip-flop-not-flip.pdf](https://github.com/bespoke-silicon-group/basejump_stl/files/3797475/flip-flop-not-flip.pdf)