bespoke-silicon-group/basejump_stl #174: [bsg_mem_1r1w_sync_synth] add feature "latch last read data"

gaozihou, 2019-10-11:
Similar to bsg_mem_1rw_sync_synth, may add parameter latch_last_read_p to control whether to hold last read value when r_v_i goes low.

dpetrisko, 2019-10-11:
I actually looked at this a bit back. There’s a little complication with latch_last_read for 1r1w. Basically, you need to handle read/write  same address cases with your latched read. The design I came up with made it so latch_last_read implied read/write forwarding, but there might be a more efficient way

gaozihou, 2019-10-11:
For current code (https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_mem/bsg_mem_1r1w_sync_synth.v) read write same address is allowed. Seems that adding a bsg_reg_en_bypass will not affect this ability?

taylor-bsg, 2019-10-11:
But in the synthesizeable version it does not matter so much if you do
simultaneous read/write. In the 1R1W synchronous hard version it could
result in invalid reads or in totally corrupted ram.

taylor-bsg, 2026-09-11:
| 7 | Issue [#174](https://github.com/bespoke-silicon-group/basejump_stl/issues/174) — [bsg_mem_1r1w_sync_synth] add feature "latch last read data" | Close — implemented | The synchronous 1R1W memory and its synthesis implementation now accept latch_last_read_p and implement last-read retention. [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/bsg_mem/bsg_mem_1r1w_sync_synth.sv#L84). |

closed 2026-09-11 by taylor-bsg