bespoke-silicon-group/basejump_stl #175: [bsg_mesh_router] bsg_mesh_router_dor_decoder in separate file?

tommydcjung, 2019-10-11:
https://github.com/bespoke-silicon-group/basejump_stl/blob/4e1931845e113de79c013644f4bf30f780246744/bsg_noc/bsg_mesh_router.v#L7

in support of one-file-one-module policy?

taylor-bsg, 2019-10-11:
Basically the expectation is that nobody will ever instantiate this module
by itself, but SV does not support private module instantiaton.

taylor-bsg, 2026-09-11:
| 11 | Issue [#175](https://github.com/bespoke-silicon-group/basejump_stl/issues/175) — [bsg_mesh_router] bsg_mesh_router_dor_decoder in separate file? | Close — implemented | The mesh DOR decoder is now in its own bsg_mesh_router_decoder_dor.sv file. [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/bsg_noc/bsg_mesh_router_decoder_dor.sv). |

closed 2026-09-11 by taylor-bsg