bespoke-silicon-group/basejump_stl #179: [bsg_mul_pipelined] clock_i port used for submodule 'bsg_dff_en.v'

tanglingshu, 2019-10-20:
In bsg_misc/bsg_mul_pipelined.v, clock_i is given to the submodule 'bsg_dff_en'. However, bsg_dff_en.v uses clk_i instead of clock_i.

taylor-bsg, 2019-10-22:
Feel free to submit a pull request

taylor-bsg, 2026-09-11:
| 10 | Issue [#179](https://github.com/bespoke-silicon-group/basejump_stl/issues/179) — [bsg_mul_pipelined] clock_i port used for submodule 'bsg_dff_en.v' | Close — specific port bug fixed | The bsg_dff_en instances in bsg_mul_pipelined now use clk_i. Keep the separate bsg_mul wrapper port fix in #693; that is a different call site. [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/bsg_misc/bsg_mul_pipelined.sv). |

closed 2026-09-11 by taylor-bsg