bespoke-silicon-group/basejump_stl #187: The vscale submodule ceased to exist

tgorochowik, 2019-11-12:
Hello,

I am creating this issue to let you know, that the vscale repository (https://github.com/ucb-bar/vscale) which is used as a submodule under `bsg_riscv/imports/vscale` no longer exists (at least not under the same address)

dpetrisko, 2019-11-12:
Thanks for the issue!  It seems like that repository has completely vanished from the web.  I wonder if it's moved as part of the ChipYard migration that Berkeley is doing?  We'll try to track it down and if not available, remove the broken submodule.

taylor-bsg, 2019-11-12:
Is there something in bsg_riscv that actually uses vscale? Vscale was buggy
anyways

dpetrisko, 2019-11-12:
Not actually used anywhere.  Closest thing is we have bsg_vscale_pkg used in hasti converter.  I'll make a PR removing vscale

closed 2019-12-14 by taylor-bsg