bespoke-silicon-group/basejump_stl #193: clean up bsg_idiv

taylor-bsg, 2019-11-22:
https://github.com/bespoke-silicon-group/basejump_stl/blob/91c5ab0e37ce868bb71a0616298cc1c70733c668/bsg_misc/bsg_idiv_iterative.v#L46

clean up the module to remove buffering of datapath signals, and structural instantiation of single-bit control registers.

dpetrisko, 2019-11-22:
> structural instantiation of single-bit control registers.

What's the style guide ruling on this? Control registers in sequential logic blocks, structural datapath?

taylor-bsg, 2019-12-03:
Open to suggestions, I think the style guide ruling is basically that you
should have a schematic for your hardware and that the relationship between
between the schematic and your code should be clear.  We don’t have any
specific requirement that you structurally instantiate anything instead of
writing RTL, but hierarchy should exist around logic that needs to be
placed in a particular place, or on datapath blocks where critical paths
are likely to approach or exceed 30 FO4. Async reset flops should be
instantiated structurally. (Hoping that I am not regretting this relaxing
of the rules.)

closed 2021-08-06 by taylor-bsg