bespoke-silicon-group/basejump_stl #197: Example of bsg_mesh_stitch with multiple networks

dpetrisko, 2019-12-01:
Can't figure out how to use bsg_mesh_stitch to stitch together multiple networks.  5-D arrays are confusing -- an example would be helpful.

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_noc/bsg_mesh_stitch.v

taylor-bsg, 2019-12-01:
https://github.com/bespoke-silicon-group/bsg_manycore/blob/master/v/bsg_manycore.v

dpetrisko, 2019-12-01:
That only has nets_p=1. I’d like to combine multiple networks with nets_p=2+

taylor-bsg, 2019-12-03:
In the manycore, it ended up being more convenient to put the plurality of
networks in a struct in the innermost array scope rather than the
outermost; for example bsg_manycore_link_sif_s in
https://github.com/bespoke-silicon-group/bsg_manycore/blob/master/v/bsg_manycore_pkg.v.
I think one reason is that the various networks may not necessarily the
same width; the other reason is that the natural usage of these networks is
that you generally only access the different networks at the leaf nodes, so
it does not make sense to put the networks at the first indirection of the
array. This can be addressed by
https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_transpose.v
but
that is very cumbersome.

M

dpetrisko, 2019-12-04:
Interesting.  The same width point is a good one.  It seems like nets_p is useless as it currently stands then, because nets_p=1 with a struct more powerful than nets_p=2 with concatenated equal sized nets.

taylor-bsg, 2019-12-04:
I think it is potentially useful if you want to specify a parameterizable
number of networks, although probably the nets should come last in the
order and not first. But agreed that it is a deprecatable feature as it
stands!