bespoke-silicon-group/basejump_stl #200: module name should be noun_adjective, not adjective_noun

taylor-bsg, 2019-12-03:
https://github.com/bespoke-silicon-group/basejump_stl/blob/6dd02f7bf07da306620c90774b4a02825a054d87/bsg_dataflow/bsg_stable_sort.v#L15

Hi Ling can you rename this module (and its users) to be bsg_sort_stable? We put the adjective after the noun (or adverb after the verb) so that related modules will appear together in an alphabetical list.

taylor-bsg, 2019-12-03:
@tanglingshu

tanglingshu, 2019-12-14:
Issue fix is under pull request #208 

However, I suggest other changes such as:

1) remove bsg_sbox_ctrl.v (it is a duplicate of bsg_sbox_ctrl_concentrate.v)
2) remove bsg_sort_4.v (it specifies items_p=4, but can be generalized by using bsg_sort_stable.v)
3) remove bsg_stable_sort.v since it will now be replaced by bsg_sort_stable.v

taylor-bsg, 2019-12-15:
Thanks! Maybe do another pull request with these changes?

taylor-bsg, 2020-01-10:
@tanglingshu

dpetrisko, 2020-06-15:
Closed with #208

closed 2020-06-15 by dpetrisko