bespoke-silicon-group/basejump_stl #209: bsg_trace_replay + bsg_trace_rom methodology does not scale well with Verilator

dpetrisko, 2019-12-15:
This is an issue with a few possible solutions, feedback welcome.

In Verilator, compiling a modestly sized bsg_trace_rom (1000) rows takes a minute.  A larger rom (10000+) takes nearly half an hour.  The same roms are compiled almost instantly in VCS.

A few spitball ideas:
1) Try to patch verilator to compile roms faster, since they should be just a giant array
2) Use readmemh to populate contents of ROM
3) Add a bsg_nonsynth_rom which dynamically queries its entries from disk via DPI, or similar

taylor-bsg, 2019-12-15:
Have we filed a bug against Verilator?

M

dpetrisko, 2019-12-18:
Not yet, since this is a performance bug, we'll need some reproducible experiments to back up the claim.  (someone else feel free to file)