bespoke-silicon-group/basejump_stl #210: Regarding bsg_two_fifo deq_i signal

saptadeeppal, 2020-01-03:
I think the deq_i signal should not depend on yumi_i alone but both yumi_i and ~empty_r. 

In the current implementation, if yumi_i is high, deq_i is high even if it's empty and the head keeps flipping. In a certain case, the valid data can be stored in a different memory element and read from a different one. Am I right or missing something here?

dpetrisko, 2020-01-03:
Hi, thanks for the issue. 

The output handshake is valid _then_ yumi. The sequence is the fifo raises valid when it is not empty, the consumer examines that signal and then raises yumi when it is ready to accept. From both the consumer side and the producer side, the transaction is complete when yumi is high. Because yumi depends on valid and valid depends on empty, your suggestion is implicitly already in the code. 

> In a certain case, the valid data can be stored in a different memory element and read from a different one.

Could you elaborate on this?

saptadeeppal, 2020-01-03:
Is it, I think the ready_o signal isn't dependent on v_i and only depends on ~full_r. So the consumer sends out ready out as soon as it can take something.    

Explanation of the case I was talking about is as follows:
Producer is initially empty and consumer is empty, so yumi that producer sees is asserted. Therefore, the deq_i is asserted and is flipping the head_r pointer every posedge clk. Now when the producer receives data it goes in to mem[0] as tail_r is at 0 and it asserts v_o. At this point if head_r = 1, consumer will read the data from mem[1] which would be wrong.

dpetrisko, 2020-01-03:
> Is it, I think the ready_o signal isn't dependent on v_i and only depends on ~full_r. So the consumer sends out ready out as soon as it can take something.
> Producer is initially empty and consumer is empty, so yumi that producer sees is asserted

I think you are conflating the handshakes of the two sides of the FIFO.

On the input side, the FIFO is a helpful consumer and there is a ready-and-valid handshake.  The signals associated are **ready_o, v_i, data_i**.  FIFO asserts **ready_o** when it is ready to accept data at the beginning of the cycle (when ~full).  The producer asserts **v_i** when **data_i** is valid.

On the output side, the FIFO is a helpful producer are there is a valid-then-yumi handshake.  The signals associated are **v_o, data_o, yumi_i**.  FIFO asserts **v_o** when **data_o** is valid (when ~empty). The consumer inspects **v_o and data_o** and then asserts **yumi_i** when it is ready to accept the transaction.

taylor-bsg, 2020-01-03:
Saptadeep, it may be helpful for you to review this paper:
http://cseweb.ucsd.edu/~mbtaylor/papers/Taylor_DAC_BaseJump_STL_2018.pdf to
understand the handshakes used in BaseJump STL. The FIFO input is rv->& and
the FIFO output is v->r.

closed 2020-01-13 by taylor-bsg