bespoke-silicon-group/basejump_stl #219: bsg_wormhole_router_buffered?

dpetrisko, 2020-01-17:
We're seeing feedthrough paths coming from bsg_wormhole_router, link_i.ready_and_rev -> link_o.v

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_noc/bsg_wormhole_router_output_control.v#L48

For timing considerations, this handshake should be ready-valid-and.  If it's not possible to change output_control to be helpful, then I think we need a buffered version of bsg_wormhole_router

dpetrisko, 2020-01-17:
I think 

```
---assign valid_o = ready_i & (|(scheduled_n & valid_i));
+++if (ready_THEN_valid)
+++  assign valid_o = ready_i & (|(scheduled_n & valid_i));
+++else
+++  assign valid_o = (|(scheduled_n & valid_i));
```

similar to bsg_two_fifo should work, provided we also make a similar changes to bsg_wormhole_input_adapter / bsg_wormhole_output_adapter

gaozihou, 2020-01-17:
Chun also noticed this path when pushing through the back-end and discussed with me, our temporary solution was simply removing "ready_i & " on valid_o assignment and it worked. My bad that I did not create an issue for this...

I think the interface implies that this should be rv&, we should be able to remove "ready_i & " safely.

dpetrisko, 2020-01-17:
Oh perfect!

dpetrisko, 2020-01-17:
Yes, I think you're right about the interface.  Looks like if you follow the path into the adapters, they are rv& fifos

taylor-bsg, 2020-01-21:
To be clear Paul's proposed solution has two parts:

1. removing ready_i on computation of valid_i
2. updating yumi_i input to include & ready_i

Paul can you run the wormhole router regression to test it?

taylor-bsg, 2020-01-21:
See https://github.com/bespoke-silicon-group/basejump_stl/pull/220

dpetrisko, 2020-05-10:
This was merged in #220

closed 2020-05-10 by dpetrisko