bespoke-silicon-group/basejump_stl #239: Redundancy for trace replay?

dpetrisko, 2020-02-19:
Are there any differences between bsg_fsb/bsg_fsb_node_trace_replay and bsg_test/bsg_trace_replay?  Can we consolidate?

taylor-bsg, 2020-03-12:
https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_test/bsg_trace_replay.v

versus

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_fsb/bsg_fsb_node_trace_replay.v

Yeah eventually I think the plan is to obsolete bsg_fsb/*, now that we are using bsg_link and bsg_wormhole.

However the EE 477 lab still uses it, so we that would be the thing to fix first. And there is some stuff with level shifters in there that might make sense to replicate outside of bsg_fsb

taylor-bsg, 2020-09-30:
Adding annoying warning to bsg_fsb_node_trace_replay =)