bespoke-silicon-group/basejump_stl #242: [Suggestion] Package Manager like NPM or Vcpkg

Xusine1131, 2020-02-22:
Recently I'm using Basejump STL in my project. And I found two points:

1. Sometimes it might be difficult to determine the dependency of an IP core. For instance, if I want to import a FIFO, I have to check out the testbench for the dependencies, or recursively find the module it refers to. Sometimes it's quite inconvenient, especially  a large system like a cache,

2. It's quite normal that I only need one module in the Basejump library, but I have to download the whole repo.

So In my opinion, a graceful way to solve these problems is a package manager, like npm for the nodejs, or vcpkg for cpp.  To do so, we need some preparations. First, we have to maintain a configuration file for each IP where the dependency and other metadata are included. (I think PPA and test configuration for Cocotb can be integrated as well) Then a client to manage own project locally is necessary. It can be used to search, download, and update the IP cores from Basejump STL.  Support the name of the tool is `vpm` (Verilog Package Manager), it's expected to work like this:

```sh
$ vpm search fifo
Searching...
1. bsg_fifo_1r1w_large
2. bsg_fifo_1r1w_narrow
# there might be other FIFOs. 
Please select a package to download: 
```

If you select 1, vpm will download `bsg_fifo_1r1w_large` from Basejump STL and put it in a specific folder where all the modules managed by vpm are. If users want to use them in a module, they just need to maintain a configuration file recording the dependencies as well.

Before building their project, users can use vpm to generate a file list for the top design. 

This is only a simple idea, and I'm not sure it works. Open to any suggestions!

dpetrisko, 2020-02-22:
Thanks for the suggestion! The best instance of what you’re describing is called FuseSoC. We’d love someone to integrate it with BaseJump STL!

Xusine1131, 2020-02-22:
@dpetrisko Thanks! And it's a huge work if we manually integrate it with Basejump STL... 
I wonder whether there are any parsers to analyze the dependency automatically... 
I'm quite lazy. :joy:

dpetrisko, 2020-02-22:
My gut feeling is that once the first few modules are integrated, we can crowdsource and the rest will come easily. At least that’s the hope!

WenhMa, 2020-09-21:
I think FuseSoC may not be the best way for a hardware STL, since it is designed to reuse big hardware IPs rather than small hardware library cells.The main problem of Basejump STL is that it is still not a software-liked library. There is no difference for users to use the modules in Basejump STL and their own modules. A more general design methodology is needed. We have built a new hardware library as a demo. I think some of the features should be used in Basejump STL. I'm trying to discuss it with @taylor-bsg, recently. If anyone is interested in the project. Please contact me.