bespoke-silicon-group/basejump_stl #255: Misleading parameter name in bsg_counter_clear_up

dpetrisko, 2020-03-18:
https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_counter_clear_up.v#L6

Should pass in ptr_width, not max_val_p.  The counter actually has max val equal to the nearest power of two.  This can be misleading if you're depending on overflow to clear the counter

taylor-bsg, 2020-03-18:
The overflow series of counters should be used in that case...

dpetrisko, 2020-03-18:
Oh, good tip for the use case.  Still misleading.  At the very least should print a warning

taylor-bsg, 2020-03-19:
It does print a warning on overflow by default?

But I guess a comment to clarify the usage cases and pointing to the
overflow versions is in order?

Did you have a specific usage where you ran into the problem? That would
help understand the issue.

M

dpetrisko, 2020-03-19:
Oh, I meant a warning if max val is not pow/2.  I guess in my head I equate overflow with wrap around, but that’s not necessarily true. In this counter, overflow just means “oops, we’re in uncharted territory here”. 

Comment sounds fine.  My use case would have been solved by the overflow counter, so a pointer to that would be useful.