bespoke-silicon-group/basejump_stl #294: DRAMSim3 Broken Submodule

stdavids, 2020-05-28:
When running `git submodule update --init`, I get the following message and it fails to clone DRAMSim3:

```
$ git submodule update --init
Submodule 'DRAMSim3' (git@github.com:bespoke-silicon-group/bsg_dramsim3) registered for path 'imports/DRAMSim3'
Submodule 'imports/ramulator' (https://github.com/bespoke-silicon-group/ramulator) registered for path 'imports/ramulator'
Skipping submodule 'imports/DRAMSim3'
Cloning into '/mnt/users/ssd0/homes/stdavids/SDH/bsg_bladerunner/basejump_stl/imports/ramulator'...
Submodule path 'imports/ramulator': checked out '3582c445bab9b6be52e126338f8ca676f4fc2e34'
```

mrutt92, 2020-05-28:
Please go into imports and run `make DRAMSim3` - it should clone it then.

dpetrisko, 2020-05-28:
#269

stdavids, 2020-05-28:
Should we just remove it as a submodule then? Presumably, this is skipped because it is a private repo and we don't want to break recursive checkouts?

stdavids, 2020-05-28:
Ah okay, #269 would address this if we moved to a public fork. Closing as a duplicated issue.

closed 2020-05-28 by stdavids