bespoke-silicon-group/basejump_stl #301: Add virtual channels adapter module

taylor-bsg, 2020-06-16:
Logically, bsg_channel_tunnel multiplexes multiple channels (not just
wormhole). But bsg_channel_tunnel is optimized for offchip communication,
where is this one is optimized for on-chip communication, where you have
the luxury of more wires but the expectation of lower latency.

The standard Synthesis Lectures on Network On Chip has a standard but complex version of virtual
channels. Here we propose a simpler implementation.

The module interface for bsg_wormhole_router is the following:

(input clk_i
,input reset_i
// Traffics
,input [dirs_lp-1:0][`bsg_ready_and_link_sif_width(flit_width_p)-1:0] link_i
,output [dirs_lp-1:0][`bsg_ready_and_link_sif_width(flit_width_p)-1:0]
link_o
// My coordinate data
,input [cord_markers_pos_p[dims_p]-1:0] my_cord_i
);

The wormhole router uses this structure to communicate between neighbor
routers:

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_noc/bsg_noc_links.vh

`define declare_bsg_ready_and_link_sif_s(in_data_width,in_struct_name) \
typedef struct packed { \
logic v; \
logic ready_and_rev; \
logic [in_data_width-1:0] data; \
} in_struct_name

Imagine, then for the virtual channel wormhole router, that the new
interface looks like this

`define declare_bsg_ready_and_link_vc_sif_s(in_data_width,in_struct_name,num_vc)
\
typedef struct packed { \
logic [num_vc-1:0] v; \
logic [num_vc-1:0] ready_and_rev; \
logic [in_data_width-1:0] data; \
} in_struct_name

So simply put, it is sharing the data wires, but the other wires are not
shared.

Then, to implement N virtual channels; one could build a module that has
the above interface and instantiates N bsg_wormhole_routers.  Then one needs
a module that connects the bsg_ready_and_link_sif_s of the output channels
of those routers to a single bsg_ready_and_link_vc_sif_s, but using a
bsg_mux_one_hot to select from the data of each of the wormhole routers and
multiplex it down on to one data bus. Then, it only remains to have a
module that adjusts the ready and v control signals going in each direction
on the output side. You can have a round robin arbiter to select which of
the transactions should take place. You can use that to mask the valid and
ready signals going in each direction. So that each channel only has one
valid signal  going out and one ready signal going in. On the input side of
the channels, you will have already masked the appropriate valid signals,
so you can run the signals straight into the input FIFOs, and replicate the
data wires.

In terms of the analysis of this, the standard virtual channels has V channels and N directions; would have N 1xV muxes in front of the virtual channel FIFOs, plus a NxN crossbar, plus VN fifo elements, for a total of NxV + NxN = NV+NN area +VN FIFO elements. This approach has V NxN crossbars, plus  N Vx1 muxes, for a total of  VNN+NV area + VN FIFO elements.  So this is only really advisable for small V.

taylor-bsg, 2020-06-16:
Note: in a 1D module, N=2 so more VC's might make even more sense.

dpetrisko, 2020-06-16:
Is this approach fundamentally different than having a (N+V) dimension router, where the V channels are connected?

dpetrisko, 2020-06-16:
Is this approach fundamentally different than having an (N+V) dimension router, where the V directions are connected appropriately? I suppose the external interface would have more data wires, but it's just a matter of where the mux is

taylor-bsg, 2024-09-14:
In this implementation, you could instantiate the same arbiter on both sides so you can reconstitute what channel was selected. One issue is that the sender must choose which data to send without looking at the ready in lines.  If this is a ready_and_valid protocol, it would have to decide based on the previous cycle's ready value, potentially resulting in a bubble. But if the router was using a credit protocol, then it could just use credits at the cost of an extra element of buffering.