bespoke-silicon-group/basejump_stl #302: Issue with getting bsg_riscv master on bitbucket

yunusdawji, 2020-06-16:
Hi,

Sorry for contacting you here about bitbucket repo. I am trying to use bison chip for my phd thesis and I was trying to get code from https://bitbucket.org/taylor-bsg/bsg_riscv . When i run make checkout-all, it fails as some of the rocket chip repos have disappeared. Would you be able to provide a archived version of the repo as you did in the bsg-celerity branch.

Thanks a lot in advance.

Yunus

taylor-bsg, 2020-06-17:
Dustin or Scott does this ring a bell?

stdavids, 2020-06-17:
We created backups of all the repositories at some point which can be found at https://bitbucket.org/taylor-bsg/workspace/projects/CEL. It seems that they were created after the latest master commit inside bsg_riscv. Hopefully, you can point any missing repositories to the repo of the same name in that workspace and it should work.

yunusdawji, 2020-06-17:
Thanks for the quick reply. I will give this a try.

yunusdawji, 2020-07-02:
I am trying to synthesize alpaca and post-synth rtl is throwing error due to invalid address. I was wondering if you had to do something special (such as initializing regs) for the post synthesis to work? I am already setting all the srams to random.

taylor-bsg, 2020-07-02:
Thanks for your question -- did you try coyote? That one should definitely
work...

M

yunusdawji, 2020-07-02:
Thanks for the quick reply. I see there are more goodies in coyote folder. I will try that out.

yunusdawji, 2020-10-02:
I had a question regarding clock in simulation setup. In one of the documents, I found that CORE_CLOCK_PERIOD_PS is used as the core clk peroid while IO_CLOCK_PERIOD_PS is used as the io clock period. But in the VCS setup they seem to switched. The PLL_CLK_i is used for io while io clock is used for core. Is this just a typo or am i missing something?

closed 2020-10-03 by yunusdawji