bespoke-silicon-group/basejump_stl #339: Lack of asynchronous DFFs

WenhMa, 2020-09-21:
There are no asynchronous DFFs in Basejump STL. These cells are widely used in many cases. I think they are necessary for Basejump STL.

dpetrisko, 2020-11-23:
Hello! To clarify, you mean flops with asynchronous reset?

WenhMa, 2020-11-25:
Yes. All DFFs in Basejump Stl are synchronous. But, when I discuss the DFF type with many engineer, they thought that DFFs with asynchronous reset are prefered in many cases.

taylor-bsg, 2026-09-11:
| 5 | Issue [#339](https://github.com/bespoke-silicon-group/basejump_stl/issues/339) — Lack of asynchronous DFFs | Close — implemented | bsg_dff_async_reset provides a parameterized vector DFF with asynchronous reset and configurable reset value. [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/bsg_misc/bsg_dff_async_reset.sv#L3). |

closed 2026-09-11 by taylor-bsg