bespoke-silicon-group/basejump_stl #348: Clock setup question in vcs testbench and for tapeout.

yunusdawji, 2020-10-02:
I had a question regarding clock in simulation setup. In one of the documents, I found that CORE_CLOCK_PERIOD_PS is used as the core clk peroid while IO_CLOCK_PERIOD_PS is used as the io clock period. But in the VCS setup they seem to switched. The PLL_CLK_i is used for io while io clock is used for core. Is this just a typo or am i missing something?

taylor-bsg, 2020-10-03:
Thanks for your question — maybe you can provide a link to the file/doc in
question?

M

yunusdawji, 2020-10-03:
So here in this file https://bitbucket.org/taylor-bsg/bsg_designs/src/master/toplevels/bsg_two_bison_accum/v/bsg_chip.v, you can see that core_clk_i is assigned to misc_L_4_i_int and io_master_clk_i is  assigned to PLL_CLK_i_int.

And then looking at the testbench https://bitbucket.org/taylor-bsg/bsg_rocket/src/master/common/vsrc/test_bsg.v on line 244 and 245 where we instantiate bsg_chip  p_PLL_CLK_i is assigned core_clk_i and p_misc_L_4_i is assigned to io_clk_i. **It seems that two have clocks have been flipped between these two files(bsg_chip.v and test_bsg.v).**

One last thing in https://bitbucket.org/taylor-bsg/bsg_rocket/src/master/common/vsrc/rocketTestHarness.v i see that  io_clk and core_clk are being correctly instantiated. 
bsg_nonsynth_clock_gen #(.cycle_time_p(`CORE_CLOCK_PERIOD_PS)) cc (.o(clk));
 bsg_nonsynth_clock_gen #(.cycle_time_p(`IO_CLOCK_PERIOD_PS)) ioc (.o(io_clk));

yunusdawji, 2020-10-06:
@stdavids Are you aware of this issue? Thanks in advance

taylor-bsg, 2020-10-06:
Hi Yunus,

misc_L_4_i and PLL_CLK_i_int are physical bumps on our open source BGA
package, and actual PCB traces on the open source board. So we retain their
names even though they have been repurposed. Those two traces have
different properties with regards to noise (PLL_CLK_i is super shielded),
so we may have want to have the better quality clock for I/O in our earlier
chip designs. All of our designs actually use internally generated clocks
that work pretty well, so in practice it didn’t matter which one we used.

M

yunusdawji, 2020-10-06:
Ok! Great thanks a lot for your help. Also our chip might be 2.5x2.5mm. We still want to use UCSD BGA 332 package. Do you think it will work or the package will need redesign? Also are there alternatives for this?

stdavids, 2020-10-06:
I think UCSD BGA 332 is probably usable, although you might not be able to fit all of the IO drivers in 2.5mm. The physical package itself (ie. the PCB substrate between the PCB motherboard and the silicon die) is not something that is easily tweakable from my experience. The IO pad ring on silicon can be adjusted/repurposed, we have a diagram that shows the IO's for 3mm here http://bjump.org/basejump_socket_352.html. It is possible that you might be able to squeeze the 3mm into your pad ring depending on the dimension of your GPIOs otherwise you'll need to cut a few more IOs to get it down to 2.5mm. I would also caution that you are going to be changing the bond angles so I would advise double-checking your bond angles with the packaging house to ensure you can actually package your final design.

As for the different clock speeds, bsg_comm_link will be performing a clock-domain crossing (CDC) so that IO data can be used by the core clock domain. This allows us to run the core logic (like processors or other compute) at a fast clock speed for better performance than the IO which is generally locked to a few hundred MHz.

yunusdawji, 2020-10-06:
Awesome! thanks a lot. I think this will be very helpful. I will let u know how it goes with a smaller chip

taylor-bsg, 2020-10-07:
Hi Yunus,

BasJump Socket 355 is a better fit for this die size since it targets a 3x3
chip — you may want to check to see if it meets your needs.

M

yunusdawji, 2020-10-08:
Thanks for reply! I  will checkout 355 socket, I will let u know how it goes. Our tech allows for pads with 20&40 um pitch so it might be possible to fit it in a smaller chip.

yunusdawji, 2020-11-01:
I have question regarding synthesis. I should be able to fix this but I hitting a brick wall. I am using DC for synthesis. It seems that DC is optimizing bsg_sync_sync for bsg_comm_link. Did you need any special directives to prevent this from happening?

taylor-bsg, 2020-11-04:
What node are you targeting? In the specific case of the synchronizers, I
suggest you create a corresponding directory inside basejump_stl/hard; e.g.
https://github.com/bespoke-silicon-group/basejump_stl/blob/master/hard/gf_14/bsg_async/bsg_sync_sync.v
and substitute in the corresponding files.

yunusdawji, 2020-11-04:
Thanks for the reply. We are planning to use GF-22nm. I dont think the have SYNC flipflops in std lib. Just the SDFF. I added dont_touch and it doesnt seem to remove the flip flop now but do u have experience with GF-22nm std lib?

yunusdawji, 2020-11-04:
Do you I need some special constraint in DC as we have two clock domains and synchronizers?

taylor-bsg, 2020-11-05:
Hi Yunus,

Every modern node has these — they are specially engineered to minimize
meta stability. For GF they are most likely in an add-on library that your
MPW vendor can provide or that you may not have downloaded or unpacked. I
would guess the cell has the string “SYNC2SDFFQX2” in the name.

taylor-bsg, 2020-11-05:
You can find some sample CDC constraints here (see also the parent
directory):

https://bitbucket.org/taylor-bsg/bsg_designs/src/master/toplevels/common/bsg_chip_cdc.constraints.tcl

yunusdawji, 2020-11-05:
Hi Michael,

Thanks for your help. I will check the vendor if they have these cells and If I am missing something. Also, thanks for the constraint files. I am suspecting some synthesis issues with bsg_comm_link.

Yunus

yunusdawji, 2020-11-06:
Also I am trying to debug bsg_comm_link and I wanted to run bsg_two_loopback or bsg_two_gcd. In README I found a repo (taylor-bsg/chip.git)  but I dont have access it. Would you please give me access, so I use that testbench? My bitbucket email is ydawji@gmail.com

taylor-bsg, 2020-11-08:
Hi Yunus,

That repo is a bit sensitive because of Synopsys IP, but I see the desire
to have a test bench. I suggest you take a look at bsg_link instead of
bsg_comm_link, it is what we now use in the group, and it is reasonably
well documented:

See Module 2 of http://bjump.org/vlsi

M

yunusdawji, 2020-11-08:
Hi Prof.,

Thanks a lot of this. I think this document will be very helpful. If you have time and can release just the testbench for bsg_link or bsg_comm_link that will great. Anyways, once I added constraints you have provide for bsg_chip, it fixed my post synthesis simulation issues.

Yunus

yunusdawji, 2020-11-30:
Hi Prof.,

I have a rookie question. I have been able to synthesize with bison with our accelerator. My post synthesis simulation works fine. Also my DRC and LVS is clean after pnr. But my functional verification for post pnr is failing (post pnr without sdf). Do you know why this could be? I am using the .sdc file provide in bsg_design for bsg_chip (and post synthesis simulation also looks ok).

Yunus

stdavids, 2020-12-01:
Hi Yunus, that is quite interesting that post-synth works but post pnr functional simulations fail. One step is to run a formal verification check (like formality or conformal) between the synthesis and pnr netlists. That will ensure that logically the two netlists are the same.

Beyond that, I would look into the following:

1. Double check that all delays have truly been removed (you can use the waveform and zoom all the way in). It is easy to mess this up because there are a few ways delay can be annotated (# delays and specify blocks being popular choices) and they have to be disabled separately.

2. X-propagation pessimism issues. This is caught in post-synth 99% of the time but it is worth checking. Simple way to check is having your simulator initialize all registers on boot to 0. This isn't a good idea for verification but is helpful to debug x-prop issues.

3. A last resort is finding the exact bug in a waveform. The easiest way to do this is to make sure that the logical hierarchy in the RTL/netlist is not flattened. That way you can run a post-synth and post-pnr simulation and pull up the two waveforms side-by-side and visually diff them. If the netlist hierarchy is preserved this is much easier to do (it is basically impossible without that). There are also some waveform-diff tools that work similar to the unix diff tool but since the netlists are not identical I don't think they will work out too well.

yunusdawji, 2020-12-01:
Hi Scott,

Thanks for quick reply. I did do a quick LEC check with setup provided in GF22nm pnr example project. All the nets it compares are matching but maybe i need to check this thoroughly. 

I will try the steps you outlined and see if I can find the problem. Again thanks a lot of help.

Yunus

yunusdawji, 2020-12-01:
Hi Scott,

I added delay_mode_unit to my vcs compile options and this FIXED the problem. I am did something totally opposite to removing delay but it fixed my simulation. Any ideas why this flag could help?

Yunus

yunusdawji, 2020-12-28:
Hi Scott/Michael,

I am trying to use GF22nm SRAM. I was wondering if you had any scripts for power grid for sram in gf14. GF22nm SRAM needs a very complicated power network and any example scripts would be of great help.

Thanks

Yunus