bespoke-silicon-group/basejump_stl #368: macro defines in package

taylor-bsg, 2020-12-22:
https://github.com/bespoke-silicon-group/basejump_stl/blob/d4cd62042e6f83b953a6bad75c2257f10ce0c853/bsg_test/bsg_dramsim3_pkg.v#L86

This means that the correct operation of this module will depend on its include order on the VCS command line. We should factor out into a .vh file and include appropriately.

dpetrisko, 2020-12-22:
I think packages often need to be first in the compile order regardless of defines?

mrutt92, 2020-12-23:
I'm not sure I understand the issue. What would be a situation where the macros are not defined correctly?

taylor-bsg, 2020-12-23:
Max — Macros are a preprocessor directive. They are not part of a package
even if they are placed textually inside of one. Our current approach
relies on MACRO definition leakage between files based on inclusion order,
which is brittle.

mrutt92, 2020-12-23:
I'm not sure how much these macros are used. They are pretty thin wrappers around just using the package scoping syntax. I could just remove them?

taylor-bsg, 2020-12-23:
Who wrote the macros?

mrutt92, 2020-12-23:
I did.

mrutt92, 2020-12-23:
I guess the safe thing is to just move them into their own header like you suggest.

PR #371 (2021-01-28): DRAMSim3 moves macros from source to header
Fixes #368
closed 2021-01-28 by taylor-bsg