bespoke-silicon-group/basejump_stl #372: bsg_cache should support cache block data width

Songchun-Li, 2020-12-25:
### Motivation
To implement BP Stream Protocol between L1 and L2, we would like to make L2 cache slice have parameterizable data width, ranging from dword_width_p (64 bits) to cce_block_width_p (512 bits) with 2^n. However, bsg_cache currently does not support data width equals to cache block width.

### Symptom
When data_width_p == cce_block_width_p in bsg_cache, block_size_in_words_p becomes 1, so lg_block_size_in_words_lp equals 0. And some of those signals should be 0-width, like addr_block_offset, while BSG_SAFE_CLOG2 set those signal to 1-width for now. Therefore, some signals are affected, like the data width of tag signals becomes 1 bit less mistakenly.

### Proposed solution
Relevent modules includes bsg_cache, bsg_cache_dma and bsg_cache_miss. To solve thie issue,  lg_block_size_in_words_lp will be set to 0 when block_size_in_words_p is 1. And conditional generate statement should be used for signals like addr_block_offset. Some address concat statements also need to be fixed as well.

A PR solving this issue will soon follow.

taylor-bsg, 2020-12-29:
This is a bit of weird case, that would have a lot of challenges in physical design because of the current design of the bsg_cache, but for low associativity caches, it might make sense.

dpetrisko, 2020-12-30:
>This is a bit of weird case, that would have a lot of challenges in physical design because of the current design of the bsg_cache, which reads out a data word for all sets in parallel, but for low associativity caches, or small block sizes, it might make sense.

Yeah, doubtful we'd tapeout a 512-bit word with the current bsg_cache, but the changes should be minimal and small block sizes are potentially interesting.

Long term, I envision plugging in a variant of bsg_cache that checks tags, then data.