bespoke-silicon-group/basejump_stl #420: Other tools for BSG_VIVADO_SYNTH_FAILS

dpetrisko, 2021-03-21:
https://github.com/bespoke-silicon-group/basejump_stl/blob/509b5eab91de2c1508470034cf7a3456c891cd04/bsg_misc/bsg_defines.v#L66

Yosys? OpenROAD? Would it make more sense to use `ifdef VIVADO? (no idea if that's actually a defined macro)

gaozihou, 2021-03-21:
Not sure whether this is accurate: https://forums.xilinx.com/t5/Synthesis/vivado-synthesizer-defined-macro/td-p/1076298
It says there is no pre-defined macro in Vivado other than SYNTHESIS..

taylor-bsg, 2021-03-21:
Yeah, definitely do not spent any more time looking for a Vivado macro. :-) I search extensively on the web and also grep’d Vivado binaries and source files. There are a bunch of XIL_* macros but nothing specific to synthesis. Hopefully the other synthesis tools (ie Altera) will be smart enough to define a macro. I don’t believe Yosys supports SystemVerilog, but that may change.

taylor-bsg, 2021-03-21:
Added comment.

https://github.com/bespoke-silicon-group/basejump_stl/commit/e1e2d63be656d4bcf32f966e985049d78dcde2b9

dpetrisko, 2021-03-21:
Yeah yosys is getting more SV support all the time, ultimate goal I think is to integrate a Verible front end which would take all SystemVerilog. But okay, comment is good!

closed 2021-03-21 by dpetrisko