bespoke-silicon-group/basejump_stl #428: Incorrect comments on v_i/ready_o port in bsg_one_fifo

muwyse, 2021-04-16:
The comments on the v_i/ready_o ports of bsg_one_fifo (https://github.com/bespoke-silicon-group/basejump_stl/blob/7a0596160ddc42ad08bfde78fa78cf1408f8fa23/bsg_dataflow/bsg_one_fifo.v#L25-L27) incorrectly imply that the interface is ready->valid when in reality the module implements ready&valid.

Adjusting the comments on the data_i and v_i ports to reflect this is a first step. Comments on both should be `// early` Renaming signals for clarity to v_i, ready_and_o would be ideal in the long term.

dpetrisko, 2021-04-16:
https://github.com/bespoke-silicon-group/basejump_stl/blob/7a0596160ddc42ad08bfde78fa78cf1408f8fa23/bsg_dataflow/bsg_one_fifo.v#L50

Also, this comment should read v_i, not v_o

muwyse, 2021-04-16:
> Also, this comment should read v_i, not v_o

However, if we only look at v_i, the interface becomes ready->valid since new data would be captured when the producer raises v_i, regardless of the state of ready_o (i.e., existing data will be cached if the producer does not consider ready_o before raising v_i)

dpetrisko, 2021-04-16:
https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_dataflow/bsg_fifo_1r1w_small.v#L25

Yep, we would need to add this parameter to change it