bespoke-silicon-group/basejump_stl #430: Double Trouble PCB Design

KarimHammad, 2021-04-19:
Hi,
I am creating this issue in reference to the PCB design of the double trouble board. I need to use this board in my research, and hence, I have submitted the development files [](https://github.com/bespoke-silicon-group/bsg_motherboards/tree/master/bsg_trouble/double_trouble/pcb_development) to ChinaFast PCB for a quote. The manufacturer has returned back to me with 12 components from the "BOM.csv" file as either out-of-stock or couldn't be found. Those are items# 5, 7, 14, 16, 21, 24, 31, 32, 37, 47, 55, and 57.

I did a thorough search to find a proper replacements to the above components, and I will appreciate if someone in the BSG team can confirm if the below replacements are good to go with. Here is what I found:
1- BOM Item# 5 (F951C226KSAAQ2) may be replaced by [](https://www.digikey.ca/en/products/detail/rohm-semiconductor/TCTAL1C226M8R/1209648?s=N4IgTCBcDaICoGE4EEAyBGBYwDYCyAHAEogC6AvkA)
2- BOM item# 7 (F951C106KSAAQ2) may be replaced by [](https://www.digikey.ca/en/products/detail/avx-corporation/F951C106KQAAQ2/4005234?s=N4IgTCBcDaIGIE4CsBGAwigDANgNIEUBBQ-CAXQF8g)
3- BOM item# 14 (LNJ214R82RA) may be replaced by [](https://www.digikey.ca/en/products/detail/everlight-electronics-co-ltd/EAST16083RA1/5142346?s=N4IgTCBcDaIKIEEDKAVAjANgAwA4DMASgmiALoC%2BQA)
4- BOM item# 16 (5-1814832-1) may be replaced by [](https://www.digikey.ca/en/products/detail/amphenol-rf/901-10112/3593514)
5- BOM item# 21 (151216-8422-RB) may be replaced by [](https://www.digikey.ca/en/products/detail/norcomp-inc/26641601RP2/1026673?s=N4IgTCBcDa4GxwCwEY4AZkCUAKEC6AvkA)
6- BOM item# 24 (a14081900ux0592) is not available at any vendor! could you please advise? In the meantime, I have found the following statement in "DoubleTrouble PCB Design.pdf":
_There are two jumpers: J30 and J31, which are 2.54mm pitch ones. The solution is that we ordered special 2.54mm headers for these two jumpers. No need to modify the board since they are removed in real trouble board (these are for JTAG)._ 
7-BOM item# 31 (RC1005F472CS) may be replaced by [](https://www.digikey.ca/en/products/detail/yageo/RC0402FR-074K7L/2827563?s=N4IgTCBcDaIEoGEAMAWJYBicC0SDsKA0ngDIgC6AvkA)
8- BOM item# 32 (CRA06P083100RJTA) may be replaced by [](https://www.digikey.ca/en/products/detail/yageo/TC164-JR-07100RL/1004815?s=N4IgTCBcDaICoGECMA2ALAWgFICUMAYB2JffHAGRAF0BfIA)
9- BOM item# 37 (RC1005F392CS) may be replaced by [](https://www.digikey.ca/en/products/detail/yageo/RC0402FR-073K9L/726584?s=N4IgTCBcDaIEoGEAMAWJYBicC0SDsAzANICcAMgAQgC6AvkA)
10- BOM item# 47 (M25P128-VME6TGB) may be replaced by [](https://www.digikey.ca/en/products/detail/micron-technology-inc/MT25QL128ABA1EW7-0SIT/6622726)
11- BOM item# 55 (ADP1715ARMZ-R7CT-ND) may be replaced by [](https://www.digikey.ca/en/products/detail/analog-devices-inc/ADP1715ARMZ-1-2-R7/1240281?s=N4IgTCBcDaIIIBEAKBGA7CgrHASgWQC0BaFAOjCJzRAF0BfIA)
12- BOM item# 57 (BT-150.000MBC-T) may be replaced by [](https://www.mouser.ca/ProductDetail/Epson-Timing/XG-1000CA-1500000M-CBL3?qs=Fobv33ltQGgPzRbZ2yGaoQ==)

Thanks in advance.

Karim

taylor-bsg, 2021-04-19:
Hi,

Did you check Digikey and Mouser for these parts?

KarimHammad, 2021-04-19:
> Hi, Did you check Digikey and Mouser for these parts?
> […](#)

Thanks a lot for the fast response! Yes I did but some of the above replacements have differences with their originals. For instance, the replacement capacitor mentioned above for BOM item# 5 is different from the original capacitor in terms of: ESR (2ohm for the original vs. unspecified for the replacement), tolerance (10% for the original vs. 20% for the replacement), height (1.6 mm for the original vs. 1.2 mm for the replacement), manufacturer size code (S for the original vs. AL for the replacement), and series (F95 for the original vs. TCT for the replacement). I am not sure if these replacements would negatively impact the operation of the board. That is why I wanted to make sure that all the above replacements won't affect the board's operation.

If needed, I can also write here the differences I found for each replacement with its original part.

Thanks again.

Karim

taylor-bsg, 2021-04-19:
Who is doing PCB Assembly for you? It might be worth contacting several
vendors. We recommend Sierra Proto

M

KarimHammad, 2021-04-19:
> Who is doing PCB Assembly for you? It might be worth contacting several vendors. We recommend Sierra Proto M
> […](#)

The quote request was submitted to ChinaFast PCB for both manufacturing and assembly. They didn't confirm yet about the assembly part since the above 12 BOM items need to be replaced first. But I am assuming they will be taking care of the assembly as well since they didn't mention no (I might be wrong). I would definitely ask ChinaFast PCB now about the assembly if you recommend that.

Karim

KarimHammad, 2021-04-19:
> Yeah I would talk to Sierra and see if they can source the parts. They are a US PCB house. We did not use ChinaFast PCB for parts procurement and assembly.
> […](#)

Good to know, thanks!
Sure, I will reach out to Sierra and ask them. 
So this means that I first get blank PCB boards from ChinaFast PCB and then send them to Sierra for assembly, right?

Karim

KarimHammad, 2021-04-19:
> Good to know, thanks!
> Sure, I will reach out to Sierra and ask them.
> So this means that I first get blank PCB boards from ChinaFast PCB and then send them to Sierra for assembly, right?
>
> Karim

I even see Sierra do fabrication, procurement, and assembly. I think you might meant to go all with Sierra instead of ChinaFast PCB, no?

Karim

KarimHammad, 2021-04-19:
Hi again,

The PCB team at Sierra is asking me if the board design is ITAR.

Could you please confirm?

Thanks.

Karim

taylor-bsg, 2021-04-19:
The board is not ITAR.

Yes, most of our boards we do PCB fab, part ordering and assembly all at
Sierra. It costs more money but is more convenient.

KarimHammad, 2021-04-19:
Thanks a lot, much appreciated! 

Karim

KarimHammad, 2021-04-22:
Hi,
Sierra PCB team just got back to me with the following BOM issues:
1- BOM item# 25 has a mismatch between its MPN and VPN (i.e., MPN 878311420 & VPN 3M5362-ND). Which number should I go with?
2- BOM item# 5 (MPN F951C226KSAAQ2) is out-of-stock and MPN F951C226KAAAQ2 is suggested as a replacement. Both items are similar except for the ESR (2ohm for the non-stocked vs. 1.4ohm for the replacement), size/dimension (3.2mm x 1.6mm for the non-stocked vs. 3.2mm x 1.70mm for the replacement), and manufacturer size code (S for the non-stocked vs. A for the replacement). Should I confidently go with their suggested item? 
3- BOM item# 7 (MPN F951C106KSAAQ2) is out-of-stock with no suggested replacement. However, I found MPN F951C106KQAAQ2 might be a good replacement (i.e., only the ESR is 3 ohm instead of 2 ohm for the non-stocked). Does my suggested replacement sound fit?
4- BOM item# 13 (MPN TAJY337K006RNJ) is out-of-stock with no suggested replacement. However, I found MPN TAJD337K006RNJ might be a good replacement (i.e., only the height-seated (max) is 3.10mm instead of 2.00 mm, and the manufacturer size code is D instead of Y for the non-stocked, respectively). The size/dimension is the same. Does my suggested replacement sound fit?
5- BOM item# 14 (MPN LNJ214R82RA) is out-of-stock with no suggested replacement. However, I found MPN EAST16083RA1 might be a good replacement. Both items are identical in package, color, configuration, lens style, voltage forward, current test, mounting type, features, package/case, and size/dimension. On the other hand, the differences are in the lens color, lens transparency, millicandella rating, lens size, viewing angle, wavelength dominant, wavelength-peak, supplier device package, and heigh-max. Does my suggested replacement sound fit?
6- BOM item# 16 (MPN 5-1814832-1) out-of-stock with no suggested replacement. However, I found MPN 901-10112 might be a good replacement. Both items are similar except in the body material (zinc alloy for the non-stocked vs. brass for the replacement), mating cycles (500 for the non-stocked vs. 200 for the replacement), maximum frequency (3GHz for the non-stocked vs. 4GHz for the replacement), and voltage rating (500 v for the non-stocked vs. unspecified for the replacement). Does my suggested replacement sound fit?
7- BOM item# 24 (MPN a14081900ux0592) has a **SOURCING PROBLEM**  (i.e., the MPN is not existing). The statement provided in "DoubleTrouble PCB Design.pdf" (i.e., quoted in my first post above in bullet 6) is not clear. Could you please advise on what should I provide the manufacturer for this item?
8- BOM item# 31 (MPN RC1005F472CS) is out-of-stock and MPN RC0402FR-074K7L is suggested as a replacement. Both items are identical with no differences. Hence, this issue is successfully resolved!
9- BOM item# 32 (MPN CRA06P083100RJTA) is out-of-stock with no suggested replacement. However, I found that MPN TC164-JR-07100RL that is identical to the non-stocked item with no differences. Hence, this issue is successfully resolved!
10- BOM item# 37 (MPN RC1005F392CS) is out-of-stock and MPN RC0402FR-073K9L is suggested as a replacement. Both items are identical with no differences. Hence, this issue is successfully resolved!
11- BOM item# 57 (MPN BT-150.000MBC-T) is out-of-stock with no suggested replacement. However, I found that MPN XG-1000CA 150.0000M-CBL3  might be a good replacement. Both items are similar in type, operating temperature, package, voltage, function, mounting type, package/case, size/dimension, frequenting. On the other hand, the differences are in base resonator (crystal for the non-socked vs. unspecified for the replacement), output (LVDS for the non-stocked vs. CMOS for the replacement), frequency stability (100ppm for the non-stocked vs. 50 ppm for the replacement), current supply-max (80mA for the non-stocked vs. 35mA for the replacement), height (1.80mm for the non-stocked vs. 1.20mm for the replacement), and current supply (disable)-max (unspecified for the non-stocked vs. 30mA for the replacement). Does my suggested replacement sound fit?

I will appreciate your insights for the issues listed above except for 8-, 9-, 10- which are believed to be resolved.

Thanks for your time!

Karim

taylor-bsg, 2021-04-22:
Thanks for your email. My response time probably will not be in scope, so
let me give you some guidelines:

- height differences should not matter too much unless you are near the
socket and the heat sink or package might have a clearance problem. You can
look at the photos online of the board to double check.

- ESR: I assume this is for caps. If the ESR is lower, then you should be
okay. If the ESR slightly larger, you will have reduced power integrity,
but a small difference would not matter.

- LEDs: any optical change is okay

- Pinout: if the placement of pads is a different, obviously you need to
rework the PCB

- Oscillator: if the oscillator changes from LVDS to CMOS you need to see
if the FPGA accepts CMOS (I assume this is where the oscillator connects)

- generally suggest Sierra over China PCB

- if one of the parts is wrong, you probably can desolder and replace. Just
watch out for power supply circuits that could be at the wrong voltage.

- you can google the part number to try to see what it is. And look at the
PCB pics to see if they replacement looks  appropriate.

- in an emergency, if the board does not work when it comes back because
you cannot source a part we can probably find some lying around to send you

gaozihou, 2021-04-22:
1. Please go with MPN 878311420.
2. Yes, suggested replacement is fine.
3. It is a good replacement.
4. Yes, 1mm taller is fine for this component.
5. That red LED is fine.
6. Your replacement is fine. However 5-1814832-2 (5-1814832-1's official replacement) is cheaper.
7. Use 929647-01-03-EU
11. Use 478-6153-1-ND

taylor-bsg, 2021-04-22:
>    8. Use 478-6153-1-ND

Thanks Paul!

KarimHammad, 2021-04-22:
Thank you so much for the very informative responses, much appreciated!

KarimHammad, 2021-04-23:
Hi Paul,

Could you please suggest a substitute for BOM item# 47 (M25P128-VME6TGB)?

Thanks again.

Karim

gaozihou, 2021-04-23:
This one is in stock: https://www.avnet.com/shop/us/products/micron/mt25ql128aba1ew9-0sit-3074457345645004448/
(I searched for NOR flash with 8x6 package in stock on multiple websites)

Refer to this for programming MT25Q flash with Spartan-6: https://www.xilinx.com/support/answers/70681.html

KarimHammad, 2021-04-26:
Thank you very much, that is super helpful!

KarimHammad, 2021-04-27:
Hi,

I have received the following two questions from Sierra:
-  Are there any components that cannot be washed?
-  Please confirm that BGAs U2 and U4 require ROHS Assembly.

I will appreciate your advice.

Karim

gaozihou, 2021-04-28:
All can be washed. ROHS is fine.

KarimHammad, 2021-04-28:
Great, thanks a lot!

Karim

KarimHammad, 2021-04-28:
I am currently ordering a new ZedBoard that I will be using to test the double trouble daughter board.

I have found three ZedBoards (with different prices and part numbers) out there:

ZedBoard 1 (part number 240-122): 
https://www.digikey.ca/en/products/detail/digilent,-inc./240-122/9841710?utm_adgroup=Evaluation%20Boards%20-%20Embedded%20-%20Complex%20Logic%20%28FPGA%2C%20CPLD%29&utm_source=google&utm_medium=cpc&utm_campaign=Shopping_Product_Development%20Boards%2C%20Kits%2C%20Programmers_NEW&utm_term=&productid=9841710&gclid=CjwKCAjwj6SEBhAOEiwAvFRuKKLvPXvxm1j3W5-I96iKf56pS5FperGIqbcuOzPIaRtYfXJ1Nkd1jBoCFegQAvD_BwE 

ZedBoard 2 (part number 410-248):
https://www.xilinx.com/products/boards-and-kits/1-elhabt.html

ZedBoard 3 (part number AES-Z7EV-7Z020-G):  
https://www.xilinx.com/products/boards-and-kits/1-8dyf-11.html

I am wondering which ZedBoard the BSG team has used?

Thanks.

Karim

KarimHammad, 2021-05-03:
Could you kindly confirm which zedboard of the three above was the one pointed out in "DoubleTrouble PCB Design.pdf"? 

Your assistance is greatly appreciated.

Karim

gaozihou, 2021-05-03:
Hi Karim,

Zedboard is available from 2 different vendors. 240-122 and 410-248 are from Digilent (240-122 has SDSoC software voucher). AES-Z7EV-7Z020-G is from Avnet. All of them should work.

However, I found one different in the product description, seems that Avnet's version has a voucher for node-locked Vivado Design Edition (shown here: https://www.xilinx.com/products/boards-and-kits/1-8dyf-11.html, also shown in AES-Z7EV-7Z020-G Getting Started Guide here: https://www.avnet.com/shop/us/products/avnet-engineering-services/aes-z7ev-7z020-g-3074457345635221599/) Not sure whether this is up-to-date, but if you need Vivado access then I guess this might be the right choice for you. (Digilent's version should not have Vivado voucher, as stated on their website: https://store.digilentinc.com/zedboard-zynq-7000-arm-fpga-soc-development-board/)

KarimHammad, 2021-05-03:
Hi Paul,

Thank you so much for your clarification!

I do have a Vivado access, and that makes the choice more flexible.

Since all of them will work with the double trouble, I will then order one from Digilent.

Thanks again for your fast and very assistive response.

Have a great day!

Karim

KarimHammad, 2021-05-21:
Hi Paul,

I have received the three inquiries from Sierra regarding the assembly of the double trouble board V1.2 and I will appreciate your assistance on addressing them:

Reviewing your files for production, we have encountered below issues that need to be resolved:
1) We need you to advise if the marked pad of components mentioned below is CATHODES:
D1, D3, D4, D11, D12, D13, D15, D16, D17 (LINE Item14) 
[image]

2) We need you to advise if the marked pad of components mentioned below is positive polarity:
C12, C82 (LINE Item3) [MPN: T491D107K016AT]
[image]

C2, C3, C83, C84, C102, C108, C109, C110 (LINE Item5) [MPN: F951C226KAAAQ2]
[image]

C27, C260 (LINE Item7) [MPN: F951C106KQAAQ2]
[image]

C5, C6 (LINE Item10) [MPN: F951A336KAAAQ2]
[image]

C9, C64 (LINE Item13) [MPN: TAJD337K006RNJ]
[image]

3) For below mentioned components reference designators are missing in silkscreen layer as well as in Assembly Drawing. Please confirm if we marked correct locations for these components.?
J2, J3, J27 (LINE Item18) [MPN: 951103-8622-AR]
J29 (LINE Item22) [MPN: 22-11-2032]
J4, J5, J10, J12, J14, J15 (LINE Item23) [MPN: 951102-8622-AR]
[image]

J30, J31 (LINE Item24) [MPN: 929647-01-03-EU]
[image]

Thanks in advance!

Karim

KarimHammad, 2021-05-26:
Hi Paul,

I will appreciate your insights about the above three inquires (raised by Sierra's assembly team).

Thanks for your time and assistance!

Karim

KarimHammad, 2021-07-12:
Hi,

Good news! Our assembled double trouble board has finally arrived today to our lab. 

I would like to deeply thank you for all your great help to make this happen!

Karim

[image]

taylor-bsg, 2021-07-12:
Very exciting Karim!!

M

KarimHammad, 2021-07-13:
Thanks Michael!
Indeed, we are super excited about it and can't wait to have it up and running :)
Once again, Thanks a lot for all the help your team has provided and for this great piece of art!

Karim

closed 2023-03-17 by taylor-bsg