bespoke-silicon-group/basejump_stl #446: BSG_ABSTRACT does not work for verilator toplevel

dpetrisko, 2021-06-07:
verilator --cc -pvalue+width_p=2 -pvalue+els_p=2 bsg_mux.v

```
%Warning-PINMISSING: bsg_mux.v:30: Cell has missing pin: data_i
%Warning-PINMISSING: Use "/* verilator lint_off PINMISSING */" and lint_on around source to disable this message.
%Warning-PINMISSING: bsg_mux.v:30: Cell has missing pin: sel_i
%Warning-PINMISSING: bsg_mux.v:30: Cell has missing pin: data_o
%Error: Specified --top-module 'bsg_mux' isn't at the top level, it's under another cell 'bsg_mux__abstract'
%Error: Exiting due to 1 error(s), 3 warning(s)
%Error: Command Failed /usr/bin/verilator_bin --cc -pvalue\+width_p\=2 -pvalue\+els_p\=2 --top-module bsg_mux bsg_mux.v
```

Haven't thought about this too hard. Seems like a Verilator issue, but could also be related to the spec. We could raise an issue there as well?

dpetrisko, 2021-06-07:
NOTE: This arises when using the concrete module as a toplevel in a file which has an abstract wrapper. When using it as a submodule, this is not an issue

taylor-bsg, 2021-06-07:
Yeah use a real top level module

dpetrisko, 2021-06-07:
doesn't work for linting or testing with verilator

dpetrisko, 2021-06-07:
I mean one approach is to say you can't directly instantiate basejump modules (since we want to do this abstract param thing for everything eventually), but that's not great.

My suggestion is to add an ifdef BSG_ENABLE_ABSTRACT which disables the abstract instantiation

taylor-bsg, 2021-06-07:
Where is the mux getting its inputs from?

dpetrisko, 2021-06-07:
The non-verilated testbench. One can verilate a testbench as a toplevel (this is what we do as BP), but for other testing methodologies (cocotb, systemC integration, mocking certain large modules as done in certain large chips) this is insufficient

taylor-bsg, 2021-06-08:
Does seem like a bug in verilator to me.

M

dpetrisko, 2021-06-10:
https://github.com/verilator/verilator/issues/3014

dpetrisko, 2021-06-10:
Not sure if it's a bug or we're misinterpreting the spec and other simulators/synthesizers are more generous

PR #450 (2021-06-25): Adding pinmissing lint exception for BSG_ABSTRACT_MODULE
This + https://github.com/verilator/verilator/pull/3026 fixes #446 and allows verilator to work with BSG_ABSTRACT_MODULE construct
closed 2021-06-25 by taylor-bsg