bespoke-silicon-group/basejump_stl #465: Potential bug in hardened fifo module

Shashank-Vijay, 2021-07-19:
The hardened fifo module contains a variable `read_write_same_addr_n`, which is currently assigned as `wptr_r == rptr_r`. The comment for this variable mentions that it should be 1 when the fifo is empty, but the condition is true for both empty and full fifo conditions. The line in question is given below:

https://github.com/bespoke-silicon-group/basejump_stl/blob/4cfaeb454a7278f7d13eb766dad07ce34fefcdea/bsg_dataflow/bsg_fifo_1r1w_small_hardened.v#L118

If this variable is 1 when the fifo is full, it prevents the memory read as in https://github.com/bespoke-silicon-group/basejump_stl/blob/4cfaeb454a7278f7d13eb766dad07ce34fefcdea/bsg_dataflow/bsg_fifo_1r1w_small_hardened.v#L95
The assignment can instead be `assign read_write_same_addr_n = empty`.

dpetrisko, 2021-07-19:
What's the bug?

Shashank-Vijay, 2021-07-19:
The 1r1w_sync memory is not read when the fifo is full because the `read_write_same_addr_n` value is 1. The r_v_i port is 0 when this variable is 1. https://github.com/bespoke-silicon-group/basejump_stl/blob/4cfaeb454a7278f7d13eb766dad07ce34fefcdea/bsg_dataflow/bsg_fifo_1r1w_small_hardened.v#L95

svenka3, 2021-07-19:
@Shashank-Vijay I didn't simulate that case, but usually we add a bypass condition and this code also has similar logic. Please look at write_bypass*

Do you have waves or a Test showing problem (if any)? 

Regards
Srini

dpetrisko, 2021-07-19:
I can confirm this bug by modifying the testbench as so:

```
diff --git a/testing/bsg_dataflow/bsg_fifo_1r1w_small_hardened/bsg_fifo_1r1w_small_hardened_tester.v b/testing/bsg_dataflow/bsg_fifo_1r1w_small_hardened/bsg_fifo_1r1w_small_hardened_tester.v
index 6244f261..dc3740da 100644
--- a/testing/bsg_dataflow/bsg_fifo_1r1w_small_hardened/bsg_fifo_1r1w_small_hardened_tester.v
+++ b/testing/bsg_dataflow/bsg_fifo_1r1w_small_hardened/bsg_fifo_1r1w_small_hardened_tester.v
@@ -108,8 +108,8 @@ module bsg_fifo_1r1w_small_hardened_tester

   // Simulation of Clock
   always #4 master_clk = ~master_clk;
-  always #4 fifo_clk   = ~fifo_clk;
-  always #4 client_clk = ~client_clk;
+  always #40 fifo_clk   = ~fifo_clk;
+  always #400 client_clk = ~client_clk;

   initial
```

taylor-bsg, 2021-07-19:
@gaozihou I presume we don't actually use this in any of our tapeouts? I would assume it would fail pretty fast.

gaozihou, 2021-07-20:
Thanks a lot for everyone on finding this real bug.. I am working on fixing this module.

@taylor-bsg unfortunately we are using this module in our tapeout. It is used in two places:
1. bsg_kink_ddr_downstream, used as receive fifo (64 elements). Simulation did not fail because total buffer size was 64+8=72 (8 extra in async-fifo), so the hardened fifo never gets full (sender stops sending until token signal received).
2. bsg_channel_tunnel_in, used as receive buffer (also 64 elements). Simulation did not fail because IO clock is slower than Manycore clock, so the receive buffer did not fill up. It is possible to trigger this bug when Manycore clock is slower than IO clock. Channel tunnel is flow-controlled by credit packets, so we'll add a patch on the FPGA to pause transmission after  48 packets are sent, until receiving the credit packet from ASIC. This could prevent the FIFO from being filled out.

taylor-bsg, 2021-07-20:
Sounds like a good mitigation plan! Looking back the two points of failure are presumably 1) a lack of testbench that exercised this case; and 2) not testing I/O and core clock at different relative speeds in the chip. 

Note that we did do RTL simulation of this module, which does set the ram output data to X if read is not valid on a ram access, so that would have caught the bug if our testing had exercised this corner case.

dpetrisko, 2021-07-20:
Also, cover points of r/v interface would have shown this gap

closed 2021-10-21 by gaozihou