bespoke-silicon-group/basejump_stl #482: RealTrouble PCB Implementation

cmcjames, 2021-08-04:
We are working on a RealTrouble Implementation using the BGA_332 package.  I am focused on the PCB aspects.

Q1) I see this picture for the ASIC pins, a 22x22 array:
https://github.com/bespoke-silicon-group/bsg_packaging/blob/master/ucsd_bga_332/pcb/352_bga_pins_top_down_flat.pdf
and I can’t find the Ironwood socket to match.  I only see this Ironwood SG-BGA676 socket, which is 26x26pins.
https://raw.githubusercontent.com/bespoke-silicon-group/bsg_motherboards/master/bsg_trouble/super_trouble_v2/misc_docs/C17099%20Dwg.pdf
_What is the link to the correct Ironwood socket for us to order?  Do we need any special permission from BaseJump to have Ironwood manufacture the socket for us?_

Q2) For the ASIC, can you confirm the ‘Netlist’ tab in this spreadsheet is what we should base the pin names and descriptions on (BGA# and Net respectively).
https://github.com/bespoke-silicon-group/bsg_packaging/blob/master/ucsd_bga_332/pcb/UCSD001-DesignInfo-2014-03-15.xlsx

Q3) We need to substitute the flash chip.  The original chip is: M25P128-VME6TGB and you recommended to my colleague for the DT board MT25QL128ABA1EW9-0SIT.  The first is obsolete and the second is out of stock. I’ve asked Micron, but they haven’t returned my request for a replacement part number.
Assuming I require the following characteristics: NOR-Flash, 128Mb as 16Mx8, SPI interface, 2.7-3.6V, Surface Mount, in-stock, ROHS, (original F=54MHz, original write cycle time: 15ms, 5ms).  It seems there aren’t any Micron parts.  Heck, when a Digikey list of options shows stock, if you click on the part, the system will then state ‘no stock’ and a call to Digikey confirms there is no stock.  This leaves a supplier called “Macronix”, I do not know if their parts are fully compatible.  We can handle a footprint change at this time, but we are hoping we won’t need to change the gateway FPGA code.  The Macronix parts in stock that fit the bill are: MX25L12835FM2I-10G, MX25L12835FMI-10G, MX25L12835FZNI-10G and links to each of those come up in [this digikey search](https://www.digikey.ca/en/products/filter/memory/774?s=N4IgjCBcoGwJxVAYygMwIYBsDOBTANCAPZQDaIALAAxwDMdIAuoQA4AuUIAymwE4CWAOwDmIAL6EwFAEyIQKSBhwFiZEAA4YAVnWzmIdpx4CR4wtIDsMOQqV5CJSOTC0KcKhSasOkbnyGiEiBaFlQ2aFj2qk4grhbxEPqGvsYBZiBwMBDQ8hHKDmpxFjpeBj5%2BJoHmHrThipEqjuTUWVTWSeWppkEAtLI5CnwAro1qWky9tQNQw6MxYGGMYstAA).  Would you recommend any one of those?

Q4) C27 is a 10UF tantalum capacitor in parallel with C37 a 0.1UF capacitor by the serial port’s MAX322EEUP.  The MAX322 datasheet requires the 0.1UF capacitor, and nothing more, what is the reason the team decided to include a large tantalum capacitor as well?

Any other hints and recommendations for a RealTrouble PCB design are welcome.

Thank you,

-James

taylor-bsg, 2021-08-06:
Thanks so much for your interest! All reasonable questions. Could you file these issues independently in the repos they pertain to?

Warm regards,

M

cmcjames, 2021-08-06:
I'm learning the system, will do, and thank you!

closed 2021-08-06 by cmcjames