bespoke-silicon-group/basejump_stl #532: Hardcoded address width in bsg_cache_to_dram_ctrl

dpetrisko, 2021-12-09:
https://github.com/bespoke-silicon-group/basejump_stl/blob/c70470142d7d090c50e6909113e26e5439199555/bsg_cache/bsg_cache_to_dram_ctrl.v#L214

dpetrisko, 2021-12-09:
Additionally, the code will not compile for addr_width < 29

taylor-bsg, 2026-09-11:
| 17 | Issue [#532](https://github.com/bespoke-silicon-group/basejump_stl/issues/532) — Hardcoded address width in bsg_cache_to_dram_ctrl | Close — superseded by rewrite | The bridge now uses dma_addr_width_p and separate raw-address/ID outputs. The reported hardcoded address concatenation was removed by the controller-interface rewrite. [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/bsg_cache/bsg_cache_to_dram_ctrl.sv#L15); [PR #556](https://github.com/bespoke-silicon-group/basejump_stl/pull/556). |

closed 2026-09-11 by taylor-bsg