bespoke-silicon-group/basejump_stl #558: Add comment that describes difference between this SIPO and others

taylor-bsg, 2022-03-01:
https://github.com/bespoke-silicon-group/basejump_stl/blob/63a7c42a5611162ceab11c91fa80d52db2d3259c/bsg_dataflow/bsg_serial_in_parallel_out_dynamic.v#L3

I think the difference is that this SIPO allows the sender to determine the number of words that should be dequed together?

dpetrisko, 2022-03-01:
> https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_dataflow/bsg_serial_in_parallel_out.v
- sender sends fixed number of flits
- receiver notified on any received
- reciever dequeues variable number of flits
> https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_dataflow/bsg_serial_in_parallel_out_full.v
- sender sends fixed number of flits
- receiver notified on all flits received
- receiver dequeues all flits at once
>https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_dataflow/bsg_serial_in_parallel_out_dynamic.v
- sender sends variable number of flits
- receiver notified on flits == sender variable
- receiver dequeues all flits at once
> https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_dataflow/bsg_serial_in_parallel_out_passthrough.v
- same as bsg_serial_in_parallel_out_full but with zero latency

taylor-bsg, 2026-09-11:
| 14 | Issue [#558](https://github.com/bespoke-silicon-group/basejump_stl/issues/558) — Add comment that describes difference between this SIPO and others | Close — documentation added | The header now explains the variable ratio, first-word output count, and difference from the ordinary SIPO. [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/bsg_dataflow/bsg_serial_in_parallel_out_dynamic.sv#L4). |

closed 2026-09-11 by taylor-bsg