bespoke-silicon-group/basejump_stl #566: Questions regarding getting to high frequency on bsg_link/bsg_comm_link

yunusdawji, 2022-04-17:
Hi,

As you know we recently taped out a chip with your libs. I see in the readme that bsg_link can reach up to 1.2 Gbps. I was wondering what type pads were used for this measurement? The I/O lib we had was working only till 150 MHz. Did you use special pads to reach such a frequency?

BR

Yunus

gaozihou, 2022-04-20:
Hi @yunusdawji, thanks for your interest in the bsg_link!

The frequency numbers were measured with regular GPIO cells provided by the manufacturer. We used the IBIS models and [simulated with these cells](https://github.com/bespoke-silicon-group/basejump_stl/tree/testing_bsg_link_io_model/testing/bsg_link/io_model). and generated the waveforms above Gbps. Here is the [direct link to the simulation guide](https://docs.google.com/document/d/1RwsDy-ASVJX0_1_qeueqQ61a_6t8cToH5fdu5GdbX4w).

Usually these regular GPIO cells are rated at 200MHz (400Mbps) or so. BSG Link uses serial/parallel on-die-termination on the FPGA side to boost the performance. The main limiting factors are the drive strength and maximum slew rate of the I/O cell, which can be verified in the simulation.

yunusdawji, 2022-04-24:
Hi Gao,

Thanks for the pointer. I will try those simulations.

The GPIO cells can reach 400Mbps with 200MHz is because the link uses DDR right?

gaozihou, 2022-04-24:
@yunusdawji Yeah the clock is running at half speed for DDR links

yunusdawji, 2022-04-24:
Ok thanks! So it seems that we can reach 175MHz (350 Mbps) with 150 MHz rate GPIOs.

closed 2022-05-29 by yunusdawji