bespoke-silicon-group/basejump_stl #573: DT and RT getting stuck sometimes

yunusdawji, 2022-06-27:
Hi,

We notice that the DoubleTrouble and RealTrouble gets stuck when we running long programs or running hello world multiple times. Are you aware of such problems? We are running bison project on DT. Also we added a ILA on ASIC FPGA and we see that the rocket gets stuck waiting for a D-Cache or I-Cache tilelink request. Any pointer or suggestion will help.

Our clock frequency is also very low, so I doubt its a timing issue.

Yunus

taylor-bsg, 2022-06-27:
Hi,

Unfortunately, it does ring a bell.  Either a bug in that version of Rocket
or in our interface code… IIRC, the Rocket host interface does polling,
that is asynchronous to the Rocket execution so it is not deterministic.
Definitely not our preferred way to implement this, we ditch the polling in
BlackParrot…

M

taylor-bsg, 2022-06-27:
> Thanks for reply. Can we just swap rocket with Blackparrot? Or does it
> need special frontend server and everything? Is there a special repo for
> Blackparrot that has zedboard, gateway and DT projects?

This may be of interest:

https://github.com/black-parrot-hdk/zynq-parrot

It runs BP inside the Zynq PL.

yunusdawji, 2022-06-27:
ok thanks!

closed 2024-06-13 by taylor-bsg